Ultimaker/Cura #2239: Wont run at all VERY MAD

KurtBenner, 2017-08-10:
Version: 2.7.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "D:/2.7/build/inst/bin/cura_app.py", line 73, in <module>
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 630, in run
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 102, in __init__
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 680, in getMachineManager
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 1124, in createMachineManager
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 99, in __init__
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 349, in setActiveMachine
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Application.py", line 152, in setGlobalContainerStack
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212, in emit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\ExtrudersModel.py", line 140, in _extrudersChanged
AttributeError: 'NoneType' object has no attribute 'containersChanged'

-------------------------
Thread 0x000006c4 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001428 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001fec (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001db4 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000012d0 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002048 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
[... 77 lines cut ...]
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000015ac (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000010ac (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000002dc (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x000013d4 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook

Vandrasc, 2017-08-14:
Hi Kurt, according to the report provided by you, something went wrong when making a change in one of the extruders stack. If you don't mind, could you send us your cura directory? This will be of great help.
It is located in C:\Users\<username>\AppData\Roaming\Cyra on Win10. 
Thanks!

KurtBenner, 2017-08-14:
Sorry but I am unable to as cura refuses to launch so I cannot access that

nickthetait, 2017-08-14:
Vandrasc is asking for a few files on your computer, cura does not need to be able to run to access this. Please open your file browser and then go the the indicated directory.

KurtBenner, 2017-08-15:
According to file explorer the file "cyra" does not exist on my computer

Vandrasc, 2017-08-15:
Oh, typo :) Did you try searching for Cura then? ;) 

You can find your user directory in C:\Users\<username>, while in username go to View menu and enable the Hidden items in order to make the AppData visible, from there select Roaming where your Cura directory is located. That should work. 
Many thanks!

nallath, 2017-08-16:
You could also try to delete that directory and try again.

ianpaschal, 2018-02-27:
Since this needed more info but we never heard back, I'm closing this issue.

closed 2018-02-27 by ianpaschal