Ultimaker/Cura #2240: Cura does not read stl

padizar, 2017-08-10:
Cura does not read stl created in Kompas 3D v.17

[conts_v2.zip](https://github.com/Ultimaker/Cura/files/1216001/conts_v2.zip)
1. Cura 2.6.1-2.7 - does not read
2. Simplify3D v3-4 - read
3. Autodesk Netfabb - read

tvanraalte, 2017-08-10:
Cuba failed to load .stl file on iMac OS Yosemite 10.10.5. First I tried Cura 2.6.2 and the file "failed to load." Then I deleted Cura 2.6.2 and downloaded and installed the Cura 2.7.0 Beta. Result was "A fatal exception has occurred that we could not recover from," along with the following bug report:

Version: 2.7.0-BETA
Platform: Darwin-14.5.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 707, in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 261, in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315, in __performEmit
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 131, in calculateBoundingBoxMesh
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 596, in getBoundingBox
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 654, in _calculateAABB
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 180, in getExtents
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 255, in getConvexHullVertices
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 247, in getConvexHull
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 239, in _computeConvexHull
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 386, in approximateConvexHull
  File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21663)
  File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3879)
scipy.spatial.qhull.QhullError: Qhull error

-------------------------
Thread 0x000000061c2c2000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000000061bdbf000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000000061b838000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000000061b31f000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/numpy/lib/arraysetops.py", line 195 in unique
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 350 in uniqueVertices
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 367 in approximateConvexHull
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 239 in _computeConvexHull
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 247 in getConvexHull
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 255 in getConvexHullVertices
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 180 in getExtents
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshFileHandler.py", line 37 in readerRead
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/FileHandler/ReadFileJob.py", line 61 in run
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Mesh/ReadMeshJob.py", line 29 in run
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 125 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000000061ae1c000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
[... cut ...]
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

tvanraalte, 2017-08-10:
I tried uploading the file in .obj format instead. Same problem. The file originates here: https://sketchfab.com/models/b7908d4fb06c49fc91d88a7ef8bd86dc#download

smartavionics, 2017-08-10:
Master branch cura doesn't appear to have a problem with the STL you supplied above...

[image]

tvanraalte, 2017-08-15:
Thanks smartavionics. Unfortunately my local library space that offers 3D printing encountered the same problem so I could not print the design I intended. We tried more than one conversion program to convert these obj files from Sketchfab to stl, but were unable to open the files (whether stl or obj) in Cura 2.6.2. Stl files from other sources worked fine in Cura. I will contact Sketchfab.

tvanraalte, 2017-08-17:
Update: we opened the obj file in MakerBot, converted it to stl, and then used that stl in Cura. It worked.

Ghostkeeper, 2017-08-23:
> Cuba failed to load .stl file

Must be that damn trade embargo...

We've just updated to a newer version of Numpy-STL, the library that does the STL file loading for us. That could fix things like this. But it didn't make it to 2.7 I'm afraid, but the version after that will have the updated library.

The error @tvanraalte was seeing is not in Numpy-STL but in QHull when we compute the collision area for the object, but perhaps Numpy-STL was giving an empty mesh and that then crashed consequently in QHull.

ianpaschal, 2018-02-27:
Since this has been inactive for more than 6 months (and is likely fixed in recent versions), I'm closing this issue.

closed 2018-02-27 by ianpaschal