Ultimaker/Cura #2243: Crash during set up of custom FDM printer

DorJamJr, 2017-08-10:
Running Cura 2.7 Beta on OS X 10.12.5. Trying to set up new printer profile (custom dual extruder FDM printer). Renamed the "Set up Custom FDM Printer" to "Prusa Dual". Then selected extruder count = 2. At that point Cura crashed. Restart Cura and now crashing on restart. Cura splash screen shows "Setting up scene" when crash report appears. Tried removing Cura 2.7 and re-installing, but still crashing. Text of Crash Report follows:

Version: 2.7.0-BETA
Platform: Darwin-16.6.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 707, in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 261, in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315, in __performEmit
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 519, in _onSettingPropertyChanged
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 558, in _updateDisallowedAreas
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 481, in getUsedExtruderStacks
KeyError: '0'

-------------------------
Thread 0x000070000d1b8000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000ccb5000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000c72f000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000c22c000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000bd29000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000b826000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000b323000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000ae20000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000a91d000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000a41a000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009e94000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009991000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000948e000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008f8b000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fffd0b6c3c0 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 116 in show
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook

DorJamJr, 2017-08-11:
Further info...
Attached is crash report of initial failure when attempting to create custom FDM printer. (Earlier crash report for this issue happened after ignoring the crash caused by trying to create custom printer and then restarting.) So this crash occurs immediately upon selecting  "2" for the number of extruders. 

Version: 2.7.0-BETA
Platform: Darwin-16.6.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 490, in _onStackChanged
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 558, in _updateDisallowedAreas
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 481, in getUsedExtruderStacks
KeyError: '0'

-------------------------
Thread 0x0000700009b5e000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000965b000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009158000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008c55000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008752000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000824f000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700007d4c000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700007849000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700007346000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
[... 35 lines cut ...]

Current thread 0x00007fffd0b6c3c0 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
[... cut ...]
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

teriperi, 2017-08-14:
Hi @DorJamJr ! I've seen a behavior similar to this before, but I began with an older version of Cura. I already had Custom FDM with 1 Extruder then I upgraded to 2.7. Then I attempted to change the extruder count to 2 and that's when I get the crash. 

Did you happen to have other existing printers before you upgraded to 2.7? 

Thanks

DorJamJr, 2017-08-14:
I actually have several versions of Cura on my MacBook (15.02.1, 2.4, 2.6.0, 2.6.1, 2.6.2). All of these have both a single and a dual extrusion printer defined. When I loaded 2.7, only the single extruder printer showed up which is why I tried to create the dual version starting with the “Custom FDM Printer” option. Within Preferences -> Printers I highlighted “Custom FDM Printer” and clicked the “Activate”button. Then I clicked the “Machine Settings” button. Within the Machine Settings dialog I set “Number of Extruders” to 2 and Cura immediately crashed and presented the Crash Report.

Sounds like we have similar experiences.

maukcc, 2017-08-15:
I am on ubuntu 16.04 with cura 2.6.99 master and have the same problem. (2.6.1 stable works)
single extruder machines work, but as soon as a multiple extruder machine is loaded, Cura crashes.
here is the log when I add the custom machine:

[cura.log.txt](https://github.com/Ultimaker/Cura/files/1224685/cura.log.txt)

teriperi, 2017-08-17:
Devs see CURA-4188

DorJamJr, 2017-08-17:
Sorry... CURA-4188 ???

teriperi, 2017-08-18:
@DorJamJr Sorry for not putting context in my last comment! It seems that you have found a bug and I was able to reproduce it. I've created an internal problem report after that. The CURA-4188 comment was for the developers' reference. :D

maukcc, 2017-08-22:
also when I use 2.7 appimage it works, but then when I check definition_changes and machine_instances, they are all at "version = 2" but the extruders are at "version = 3" 
Maybe that has something to do with it ?

Ghostkeeper, 2017-08-23:
This was fixed yesterday, so it shouldn't happen in the upcoming stable release.

closed 2017-08-23 by Ghostkeeper

maukcc, 2017-09-04:
This is probably fixed in the 2.7 branch, but not yet in the master(2.6.99) branch.
As I am on the PPA of the master branch, I can not test this.
Any time frame when this will be ported to the master branch?

Ghostkeeper, 2017-09-10:
The 2.7 branch is no longer active since 2.7 released, so if it's fixed in 2.7 but not in master we must have a regression.

The commit that should fix this was here: https://github.com/Ultimaker/Cura/commit/16764f47507192f4573cf470c08963e43bf8a716

maukcc, 2017-09-11:
The problem is that the ppa is being generated from the master version on thopiekar repository, which is lagging behind on the ultimaker repository.
The updating stopped at a very unfortunate timing, so we will have to wait.