Ultimaker/Cura #2244: A fatal exception has occurred - Different than #2225

jonathanweinberg, 2017-08-11:
Hi @Vandrasc, you mentioned my pasted log being a different issue than the one raised in #2225, so I made a new issue.

I retrieved the full log from this location: `/Users/jonathanweinberg/AppData/Local/cura/stderr.log`

It's attached here: [stderr.log.txt](https://github.com/Ultimaker/Cura/files/1216954/stderr.log.txt)

Within there, there's a line that stood out to me (repeated again, then it goes critical and dies):

`2017-08-11 00:19:08,597 - ERROR - UM.Controller.getTool [141]: Unable to find CameraTool in tools`

I only modified the local LAN IP address of my Octo Print install. No other sins of editing the log!

Incidentally, why is the log being stored in `~/AppData/Local/cura/stderr.log`? That seems pretty unusual on macOS.

Ghostkeeper, 2017-08-14:
From what I've seen, this happens when none of the plug-ins can be found. The camera tool is just the first plug-in that gives an error.

> Incidentally, why is the log being stored in `~/AppData/Local/cura/stderr.log`? That seems pretty unusual on macOS.

That could be related... The log should get stored in `~/Library/Application Support/cura/<version>/stderr.log`.

In #2225 it appears that you've built Cura yourself. The locations where it searches for plug-ins are here: https://github.com/Ultimaker/Uranium/blob/d060d33a518a47e34ab15838a7708bc01e9b630f/UM/Application.py#L106
Perhaps you haven't got the correct install prefix?

jonathanweinberg, 2017-08-14:
I am not the author of #2225. I mistakenly posted my log thinking it's similar in #2225. This was a prebuilt beta download.

Ghostkeeper, 2017-08-14:
Whoops, sorry.

Well in any case, if there is no camera tool, it won't be able to set the origin, which indeed causes the original error you're getting. So the question is still why it isn't able to load the camera plug-in then...

jonathanweinberg, 2017-08-20:
I think I'll just try the trusty old nuke and pave and see what happens on a fresh install, unless there's something else I can do to grab info for anyone.

closed 2018-03-02 by ChrisTerBeke