Ultimaker/Cura #2246: Filter option for Material Profiles

Drayson80, 2017-08-11:
Feature Request: cleanup material manager/list to display only user-relevant materials and not the very long list of possible useable materials for the printer.... Would make the material profile selection more ease to use.

Option 1: Implement a checkbox per material within Material Manager to select which one should be displayed in the program GUI and which not - makes it more ease to use and cleaned up.
Option 2: if there are materials in AppData\Roaming\cura\2.7\materials, don´t display the materials of Program Files\Cura 2.7\resources\materials

Ghostkeeper, 2017-12-11:
This issue just got rejected by our project manager because it was older than 12 weeks.

ianpaschal, 2018-05-22:
This is now "in progress" thanks to the new materials rework.