Ultimaker/Cura #2254: Print surface issues with some high-polygon count STL`s

Chrisu02, 2017-08-11:
hello as already described at the forum, it seems cura has some issue when trying to print high-polygon (or microlengh ?) files , the surface will become humps and blobs.
https://ultimaker.com/en/community/50288-cura-hi-polygon-issue-with-organic-models

Maybe an option will help to define a minimum segment length  ?

sincerly, Chri

Ghostkeeper, 2017-08-24:
The Groot model takes too much memory to load for my work laptop :(

Chrisu02, 2017-08-24:
I used the "lower" Detail model with 57 Mb, not the high res which has already a very high resolution.

Chri

Ghostkeeper, 2017-08-31:
Craftware was displaying giant globs of material filling the entire build volume for some reason. But here's a screenshot of Groot's head in RepetierHost:

[image]

Looks all right there, at least.