Ultimaker/Cura #2255: Option to disable Temperature control at cura ?

Chrisu02, 2017-08-11:
Hello, as already mentioned at the forums: https://ultimaker.com/en/community/49805-disable-temperature-control-at-cura-26

It seems that there is no possibility to disable the temperature control of the hotend completely ?
(M104 and M109)

when printing lots of parts in short time it would be easier to let the printer be on his actual temperature, otherwhise you have always to wait until the printer is a the specific temperature again.

also when using Cura on non UM printers it`s sometimes easier to control the temperature manually with the printing software.

Some other Slic3rs do interpret a value of "0" as "disabled", would be good to if cura would have also such an option.

sincerly, Chri

Ghostkeeper, 2017-08-24:
There is a machine setting that determines whether there should be temperature control, but it's not exposed to the user.

As far as I understood, 0 still means disabled in Cura too... If that's not the case that might be a bug.

Chrisu02, 2017-08-24:
then it might be a bug, because when set to "0" at cura it sets the temp to zero and not disable the command.

Ghostkeeper, 2017-08-24:
Hmm, for me setting the temperature to 0 causes an assertion to fail:

    2017-08-24 12:43:44,685 - DEBUG - UM.Backend.Backend._backendLog [92]: [Backend] CuraEngine: /home/ruben/Projects/CuraEngine/src/pathPlanning/NozzleTempInsert.cpp:16: cura::NozzleTempInsert::NozzleTempInsert(unsigned int, int, double, bool, double): Assertion `temperature != 0 && temperature != -1 && "Temperature command must be set!"' failed.

That's because I'm running CuraEngine in debug mode. It's actually a known issue and currently in our top 50 issues. But I wouldn't expect it to just continue slicing; I'd expect the slicer to crash.

fieldOfView, 2017-08-24:
Fixed, see here: http://ultimaker.com/en/community/49805-disable-temperature-control-at-cura-26?page=1&sort=#reply-187798

I tested the definition with the correct setting (machine_nozzle_temp_enabled) and it works as expected.

closed 2018-03-02 by ChrisTerBeke