Ultimaker/Cura #2256: .gcode file name does not always save with prefix

ChrisRiddell, 2017-08-11:
Cura 2.7 Beta
Ultimaker 2+

When saving gcode to an SD card the UM2_ prefix is not always exported with the name of the file

eg: Cura shows UM_test but the file exported is test.gcode

I found a work around by clicking into the name and deleting the ending letter and adding it back.

smartavionics, 2017-08-14:
I think I have seen this lack of gcode file prefix and I have a vague memory that it occurred after I had switched the printer profiles whilst keeping the same model on the build plate?

ChrisRiddell, 2017-08-14:
No printer profile changes but the names where pretty long-ish though and I have only seen this happen when saving to an SD card not to any folder on the computer.

nickthetait, 2017-08-14:
This might be a duplicate of https://github.com/Ultimaker/Cura/issues/2194

smartavionics, 2017-08-14:
Yes, it could - and that hasn't been resolved, either.

Ghostkeeper, 2017-08-24:
Indeed the reproduce steps and symptoms are exactly the same, so I'll close this one as a duplicate of #2194.

closed 2017-08-24 by Ghostkeeper