Ultimaker/Cura #2257: [2.7] Connection to OctoPrint impossible

probonopd, 2017-08-11:
Connection to OctoPrint is impossible, getting (regardless of the actual API key):

> The API key is not valid.

[image]

On the console I see

```
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
QXcbConnection: XCB error: 3 (BadWindow), sequence: 947, resource id: 38311438, major code: 40 (TranslateCoords), minor code: 0
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
file:///tmp/.mount_HxmBnT/usr/bin/plugins/plugins/OctoPrintPlugin/DiscoverOctoPrintAction.qml:236: TypeError: Cannot read property 'baseURL' of undefined
```

I think it may have to do with the last error.

It used to work with 2.6.

fieldOfView, 2017-08-11:
Can you test the following url in a browser for me?
http://192.168.0.15:80/api/settings?apikey=xxxxxxxx
Note that the full API key is 32 characters. Make sure you get the whole thing.

fieldOfView, 2017-08-11:
Note that the official issue queue for the OctoPrintPlugin is here:
https://github.com/fieldofview/OctoPrintPlugin

probonopd, 2017-08-11:
Apologies, further testing on my side shows that with the original OctoPrint instance it works whereas with my [minimal OctoPrint subset](https://github.com/probonopd/WirelessPrinting) (which just accepts any API key for the moment) it currently fails because `api/settings?apikey={16 digits}` is unimplemented.

It seems like this check did not exist before, because it used to work in older versions of Cura. Oh well, I guess I will have to implement the API keys as well in my minimal OctoPrint subset. Sorry for the false alarm.

closed 2017-08-11 by probonopd

probonopd, 2017-08-11:
@fieldOfView sorry for being semi-OT, but can you please comment in the linked issue on what is the minimal subset of a response that we can get away with? Thanks.