Ultimaker/Cura #2259: In 2.7 beta, opening a saved project produces failed to load

rtideas, 2017-08-11:
Using 2.7 beta on 27" 5K iMac run-in OS 10.12.6.

I saved a project file and when I try to reload it, I get a dialed to load. I can load them as objects, but not as a project.

teriperi, 2017-08-14:
Hi @rtideas , I'm seeing the same thing but for project files with a Custom FDM printer. Is this the same for you? May I ask what printer is in your project file? Thanks

rtideas, 2017-08-14:
i3 Chinese kit. 200 x 200 x 180 build volume.

rtideas, 2017-08-14:
If I close Cura and then double-click the the .3mf project file, Cura will open it--but i has list my Post Processing (Tweak@) settings.

teriperi, 2017-08-14:
Yes! This looks similar to what I have. I've created an internal issue for it to be looked at. 

Devs please see CURA-4167

Ghostkeeper, 2017-08-25:
This bug was fixed in 2.7 for the stable release :smiley:

closed 2017-08-25 by Ghostkeeper