Ultimaker/Cura #2260: As a user with limited screen space, I would like to close the move/scale/rotate dialogs.

krispaulsen, 2017-08-12:
[image]
After clicking a button on the left, a dialog with options is displayed, but there's no way to close it without opening another dialog or deselecting the 3D object. Clicking again on the button does not close the dialog as expected, nor does pressing the ESC key.

I found I could close the dialog by deselecting the 3D object, but it reappears as I manipulate the view, right clicking to bring up the context menu, or clicking on 3D objects.

nickthetait, 2017-08-12:
As an alternative idea: Would it be helpful if the right side panel could be manually reduced in size?

fieldOfView, 2017-08-12:
[image]

Ghostkeeper, 2017-08-25:
> The internal JIRA issue CURA-1287 might even still be assigned to me ;-)

Hehe, I'm impressed that you still know the ticket number.
But no, it's unassigned and resolved as "Won't do" because no modification was made to the ticket for 3 months... Welp, time to make a new one.

Ghostkeeper, 2017-11-21:
And the new ticket also expired.

nickthetait, 2017-11-22:
:cry:

fieldOfView, 2018-05-08:
Still no way to collapse the tool panels though...

ianpaschal, 2018-05-08:
Correct but sometimes we have to say "no."