Ultimaker/Cura #2265: BUG, Cura Scale?

zerspaner-gerd, 2017-08-12:
## BUG, Cura Scale?

Why do occasionally not some decimal numbers appear at"Scale"?
[image] Cura
[image]S3D
[image]Fusion 360

Is not the case for every stl file.
Here is one where I saw it again
[Fidget Spinner v7_Farbe eins.stl.zip](https://github.com/Ultimaker/Cura/files/1220055/Fidget.Spinner.v7_Farbe.eins.zip)

Is this a bug?
How is it scaling with the percentages, which value is used here?

zerspaner-gerd, 2017-08-12:
Seems to be file sizes dependent.
Here a cube (150,123 x 150,123 x 150,123) with different quality setting exported
[cube.stl.zip](https://github.com/Ultimaker/Cura/files/1220131/cube.stl.zip)

The Medium file is displayed correctly.

Ghostkeeper, 2017-08-25:
The scaling with percentages is the ratio between the current scale and the scale of the original file.

Ghostkeeper, 2017-08-25:
The bug reproduces for me, yeah, with both the High cube and the fidget spinner.

zerspaner-gerd, 2017-08-27:
> The scaling with percentages is the ratio between the current scale and the scale of the original file.

Cura counts the wrong displayed value or the real correct value?
e.g. With cube high
Scaled 100.19% Result Cura = 150.285
Correct would be 150,408

Does not make a trustworthy impression, especially if you want to print accurately

Ghostkeeper, 2017-09-03:
You can find that out by setting the line width to 0.001mm and inspecting the bounding box of the g-code (make sure to remove brim/skirt/raft first though).

The Cube High at default 100% scale, sliced on an UMO this way, results in a bounding box of X from 27.437 to 177.563 and Y from 27.436 to 177.563. The difference between those is 150.127. So the actual slice seems to be 2µm off, which can be attributed to rounding errors.

zerspaner-gerd, 2017-09-08:
Would be nice if the ad error could be fixed. 😀

Ghostkeeper, 2017-11-21:
Yeah, I agree. But our project manager just removed this from our planning, meaning that we won't be able to spend time on this issue on Ultimaker's time.

closed 2018-11-26 by Appesteijn