Ultimaker/Cura #2266: [Regression] Stuck on slicing, relates to `bottom_reference_wall`

protist, 2017-08-12:
When attempting to prepare a model, cura is stuck on "Slicing…". The progress bar does not move. cura worked fine with b714f0ff, but the master d9e1154c fails.

On failure, the following messages are displayed.
```
2017-08-13 09:26:54,427 - DEBUG - UM.Backend.Backend._backendLog [92]: [Backend] [ERROR] Trying to retrieve unregistered setting with no value given: 'bottom_reference_wall'
2017-08-13 09:26:54,427 - DEBUG - UM.Backend.Backend._backendLog [92]: [Backend] [ERROR] Trying to retrieve unregistered setting with no value given: 'bottom_reference_wall'
2017-08-13 09:26:54,428 - DEBUG - UM.Backend.Backend._backendLog [92]: [Backend] [ERROR]
2017-08-13 09:26:54,429 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [609]: Backend quit with return code 255. Resetting process and socket.
2017-08-13 09:26:54,965 - INFO - UM.Backend.Backend._onSocketError [201]: Backend crashed or closed.
2017-08-13 09:26:54,965 - DEBUG - UM.Backend.Backend._createSocket [213]: Previous socket existed. Closing that first.
2017-08-13 09:26:54,967 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [244]: Attempting to kill the engine process
```

I performed `git bisect` and identified the regression in b43acf5b180003315c27745f800ee080417fa21a. I checked the log for this commit, but found it a little opaque.

    Revert "Revert "Revert "feat: Skin Reference wall settings (CURA-3990)"""

However, I checked out master again, added back the lines that were removed in b43acf5b180003315c27745f800ee080417fa21a, and I can slice again.

I'm running Arch Linux, with a custom FDM printer. Model is a simple stl file exported from FreeCAD. I'm happy to provide more information if required.

smartavionics, 2017-08-13:
Sounds like your backend (CuraEngine) is out of sync with the UI (Cura). I am currently using the master branch (with additions) and it's not crashing like that.

protist, 2017-08-13:
Thank you for the quick reply. You are indeed correct. Arch Linux's AUR's curaengine-git is broken and not updating correctly. Thanks for your time, and apologies for the noise.

closed 2017-08-13 by protist