Ultimaker/Cura #2268: Slicer does not see some of the model in the stl file

Diramu, 2017-08-13:
Using 2.6.2.

Slicing  models https://www.thingiverse.com/thing:2445461
file https://www.thingiverse.com/download:3932029  (T13_B2_HD_hull.stl) was incomplete.
The a model for a tank (a friend ask me to print :-), the bottom half of the model does not show up in Cura.  It worked fine using ideaMaker.

Ghostkeeper, 2017-08-25:
Looks fine to me:
[image]
Using the Thingiviewer on Thingiverse as reference, that is.

Ghostkeeper, 2017-08-25:
I do note that CloudCompare seems to also show the bottom side as missing!
[image]
Maybe something is up with the model. CloudCompare says that it removed duplicate vertices and faces, so perhaps that has something to do with it.

ianpaschal, 2018-03-02:
Not an issue with Cura. Model seems to be bad.

closed 2018-03-02 by ianpaschal