Ultimaker/Cura #2269: Can't set 0.25nozzle with 1.75 filament

kr15uk, 2017-08-13:
Can't set 1.75mm filament on custom machine with 0.25mm nozzle.
Neither 0.15mm nozzle - looks like it doesn't like 2x digits after zero dot.

macOS Sierra 10.12.6
Cura 2.6.2

fieldOfView, 2017-08-13:
Can you add steps to reproduce?

kr15uk, 2017-08-14:
1. Open Cura 2.6.2
2. Preferences
3. Printers
4. Add
5. Custom
6. Change material from 2.85 to 1.75
7. Change nozzle size to 0.25 from 0.4
8. Finish
9. Chose any other machine in the list
10. Activate
11. Go back to 0.25mm machine
12. Activate
13. Machine Settings
14. It's back to 2.85mm

If you try to change it now to 1.75 it jumps back to 2.85 without switching the machines, just close and machine settings again.

It's something about two digits after 0.

Hope this helps. Here if you need more testing.

fieldOfView, 2017-08-14:
This has been fixed in Cura 2.7. You can work around this in Cura 2.6 by setting the material diameter to 1.75 in the sidebar.

closed 2017-08-25 by Ghostkeeper