Ultimaker/Cura #2270: Filament diameter defaults to 2.85 despite being set to 1.75

probonopd, 2017-08-13:
In Cura 2.6.2, the filament diameter defaults to __2.88__ despite being set to 1.75 in the [printer definintion](https://github.com/Ultimaker/Cura/blob/2e790852ced1d0097cbc93216237deb8e0de5117/resources/definitions/renkforce_rf100.def.json#L81-L83)

```
 "material_diameter": {
   "value": "1.75"
},
```

[image]

This only becomes visible after switching to "Custom". I have wasted many hours finding the issue...

probonopd, 2017-08-13:
Update: It happens with the Prusa i3 Mk2 profile as well.

1nt, 2017-08-15:
I had change "Material  Diameter" in "Manage Printers -> Machine Settings" at 1.75 only in Cura 2.7.

nallath, 2017-08-16:
Could you check if this is still an issue with 2.7 beta?

probonopd, 2017-08-16:
At least it is _not always_ an issue in 2.7, but I cannot say yet whether it _never_ is an issue.

probonopd, 2017-08-17:
Yes, today I had the issue also with 2.7 BETA.

huelvayork, 2017-08-19:
I upgraded from 2.5 to 2.7.0-beta. "material diameter" settings in my printer definition were lost, and changed to 2.85.

probonopd, 2017-08-19:
Thanks that finally someone can confirm it, I was starting to think it's just me ;-)

fieldOfView, 2017-08-19:
If anyone can come up with steps to reproduce this in 2.7, perhaps this can be fixed.

huelvayork, 2017-08-20:
Seems that the "Material diameter" setting in the printer mangager setup is new. It wasn't there in previous versions:

Cura 2.5:
[image]

Cura 2.7 beta:
[image]

probonopd, 2017-08-20:
Here is a way to reproduce this:
1. Use a system on which _no version of Cura ever ran_ (in this test, I used a `xubuntu-16.04.1-desktop-amd64.iso` Live system)
2. Download Cura 2.7 BETA AppImage and run it
3. In the printer selection dialog, select "Renkforce RF100"
4. Without touching any settings, flip the switch from "Recommended" to "Custom"
5. Check under "Material", it is set to __2.85__

fieldOfView, 2017-08-20:
@probonopd good catch, see #2300.

probonopd, 2017-08-20:
Thanks @fieldOfView but note that even if I go to Machine settings, it is set to 2.85 there as well. Does your fix address this, too?

fieldOfView, 2017-08-20:
Yes.