Ultimaker/Cura #2273: Cura 2.6 UM2+ firmware filament grind

ChrisRiddell, 2017-08-14:
Cura 2.6 (maybe 2.7 don't think their has been a firmware upgrade included?) 
Ultimaker 2+

Since upgrading my printer to the 2.6 firmware when changing filament it grinds it and after much researching found the issue to be before it drives the filament out its rams it forwards pretty fast first hence grinding the filament.

a photo and a video showing the issue.
https://youtu.be/O8l5sp1Tbf8

[image]

Ghostkeeper, 2017-08-28:
Please post this here instead: https://github.com/Ultimaker/UM2.1-Firmware/issues

Ghostkeeper, 2017-08-28:
Indeed there hasn't been a firmware update for 2.7.

closed 2017-09-20 by Ghostkeeper