Ultimaker/Cura #2277: Crash report

Mike13815, 2017-08-15:
Figured since it's the beta, its important to submit this. I have not gotten it to work yet. 

Version: 2.7.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 707, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189, in call
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderStack.py", line 94, in _onPropertiesChanged
AttributeError: 'NoneType' object has no attribute 'definition'

-------------------------
Thread 0x00002980 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002af8 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00000338 (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00003358 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000025ec (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000025b8 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002278 (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002a54 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000008b0 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00003230 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001bdc (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000295c (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002a74 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000002cc (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000240c (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000027e0 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x00000978 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 342 in showSplashMessage
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 652 in run
  File "D:/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Vandrasc, 2017-08-15:
Hi Mike, this will be fixed in the 2.7 release candidate. As work around you can remove your user directory from C:\Users\<username>\AppData\Roaming\Cura. Thanks!

Thisismydigitalself, 2017-08-15:
I have the exact same crash/error even after removing Cura's directory from my user directory.

Ghostkeeper, 2017-08-23:
Duplicate of #2204.

closed 2017-08-23 by Ghostkeeper