Ultimaker/Cura #2278: 2.7 Beta slicing issue and temperature issue.

Tobitulpe, 2017-08-15:
Love the new theme... unfortunateley i don't get cura 2.7 to print.

I have noticed two issues. first one occures while having a custom profile cura wont start slicing at all. Workaround is to switch in between some other profiles. After switching to another profile and switch back to the first profile again you get the object sliced as desired.

2nd one has to do s.th. with the print temperatures. With my micro3d with iMe firmware i had to call a m109 command in start gcode to get the printer going in previous cura versions. Now, as your newly doing it by default in the beginning of a print i could erase this line. You also call a m104 directly before the m109. Which seems to confuse my printer, that he heats up to desired temp and then stays idle wouldn't a single m109 be sufficent?

cheers 
tobi

`;FLAVOR:Marlin
;TIME:1356
;Filament used: 0.331503m
;Layer height: 0.2
;Generated with Cura_SteamEngine 2.7.0-BETA
M104 S185
M109 S185
;Sliced at: Tue 15-08-2017 12:54:54

G90        ;absolute positioning
G92 E0 ; Zero the extrusion length...`

Ghostkeeper, 2017-08-29:
> I have noticed two issues. first one occures while having a custom profile cura wont start slicing at all.

Is there some way in which we can reproduce this? If I try this with my own custom profile (modified infill density) it slices fine. What profile were you using?

> You also call a m104 directly before the m109. Which seems to confuse my printer

If you have multiple extruders used near the start of the print, first we give the `M104` command to all extruders so that they start heating up, then we give the `M109` command to the one we're using first to ensure that it has the correct temperature. This way you can have multiple extruders heating up at the same time which saves time. We didn't make an exception for single extrusion.

You can alter the start g-code to fix this. If there is the keyword `{material_print_temperature}` in the start g-code then that keyword will get replaced by the printing temperature, and it also won't generate the default heating up commands at the start of the file any more.

Tobitulpe, 2017-08-29:
With the profile and the notslicing issue it seems independent wether its a built in profile or a custom one. Even when the starting profile is "draft quality", the system wont start slicing after I loaded an stl file. I Always have to switch back and fourth to get it slicing. (win7, nvidia 960)

The {material_print_temperature} trick seemed to work. In former i also used it but only with a {print_temperature} command which isn't working anymore. 
Unfortunateley the printer still heats up to the desired temperature and doesent go further in code. 
Is there a possibility to monitor what is going on inside? In Cura 15.** we have the pronterface UI where you can monitor which lines are executed. (Btw. i loved the functionality of pronterface ui...: buttons for moving, homing, retracting, or executing gcode by hand is quite cool. Any chance to get it back in Cura 2?)

Cheers
Tobi

Ghostkeeper, 2017-09-03:
That's been asked quite often, but I think that whatever tickets we had got deleted in the meanwhile, so I'll create a new one.

Ghostkeeper, 2017-09-03:
The profile switching thing still doesn't reproduce for me though... The stable release for 2.7 is out now. Perhaps it's worth a try?

closed 2018-11-22 by Appesteijn