Ultimaker/Cura #2282: Feature request: Last Top Layer options

amagr0, 2017-08-16:
In order to improve 3D print finish quality on the last top layer I think more options should be opened, like the caution that already exist for the Initial Layer.

For example:

Quality:
Top/Bottom Line Width -> separated the Top and Bottom Line Width
Last Top Layer Line Width

Speed:
Top/Bottom Speed -> separated the Top and Bottom Speed
Last Top Layer Speed

Travel:
Z Hop When Retracted on Last Top Layer

Ghostkeeper, 2017-08-29:
We've implemented some of this under the nomination "Top Surface Skin" in 2.7:

* Number of Last Top Layers -> Top Surface Skin Layers
* Last Top Layer Line Width -> Top Surface Skin Line Width
* Last Top Layer Speed -> Top Surface Skin Speed

All the top/bottom settings are already separated as well, such as Top and Bottom Line Width and the speed settings, etc.

There's also several other options for the Top Surface Skin available, such as line directions, patterns, etc.

Some of them are not implemented:
* Last Top Layer Height: It is impossible in our current architecture to have different layer heights in one print. One day we'll be able to do stuff like this but not for now.
* Z Hop When Retracted on Last Top Layer: Would be a good idea to implement I think.

Eganwp, 2017-09-13:
+1 Please implement the final point above: "Z Hop when retracted on last top layer". Then we can finally all have prints without the nasty Z travel scar across them at the end! 

Even at the first layer would be a good idea for those who use glass so there's no scar travel line laid across the bottom first layer.

ianpaschal, 2018-03-02:
I'm pretty sure this has been fixed. Can reopen if it was not.

Ghostkeeper, 2018-03-26:
No, these haven't been implemented yet.

Honestly Last Top Layer Height will probably never happen, or at least not until we have separate layer heights for each layer part which is a long way off for now. But the other feature request, Z Hop When Retracted on Last Top Layer, is feasible.

filipgoc, 2018-04-04:
Actually, isn't 

>  Z Hop When Retracted on Last Top Layer
already possible with the combination of

- combing set to 'no skin'
- z hop enabled

or is the difference that this combination of settings also z hops when retracting perimeters or something?

Ghostkeeper, 2018-04-26:
Yeah that would also z-hop when other retractions occur, if combing is impossible.