Ultimaker/Cura #2283: Cura 2.6.2 Does not Slice any Model

CC-Tong, 2017-08-16:
Hello,
I am currently having the problem that Cura does not slice any loaded Model, even Models which I have printed in the past.
It Loads the Part and jumps to Slicing after changed but there are no changes and Layers are not available.
Any assistance is appreciated.

nallath, 2017-08-18:
Could you provide the cura.log file?

CC-Tong, 2017-08-19:
Thank you for your reply, I had to shorten the log file:[cura.log.txt](https://github.com/Ultimaker/Cura/files/1236196/cura.log.txt)
I never had that problem before, so I deleted the last year from the log file.
Thank you in advance and in case of any additional questions please let me know.

fieldOfView, 2017-08-19:
Could you check if this file exists?
C:\Program Files\Cura 2.6\CuraEngine.exe

If not, it could be that an anti-virus application has prevented it from being installed. In that case, please see if you can put it in that location manually. You can do so by unzipping the installer. CuraEngine.exe should be in the unzipped folder. Drag it into the C:\Program Files\Cura 2.6\ folder.

CC-Tong, 2017-08-19:
The CuraEngine.exe exist in that folder.
I have retried the installation with the antivirus turned of, same result.

Ghostkeeper, 2017-08-29:
... but...

    ERROR - UM.Backend.Backend.startEngine [81]: Unable to find backend executable: C:\Program Files\Cura 2.6\CuraEngine.exe

Can you try executing that program? Just double-click on it? You'd expect a console window to briefly appear showing a bunch of text.

Ghostkeeper, 2018-01-15:
This issue has been duplicated in the meanwhile by #3000. Because #3000 has more information and a way nicer issue number I'll close this one as a duplicate (even though it was first).

closed 2018-01-15 by Ghostkeeper