Ultimaker/Cura #2285: Crash report

drdavelight, 2017-08-17:
Version: 2.7.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "D:/2.7/build/inst/bin/cura_app.py", line 73, in <module>
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 630, in run
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 102, in __init__
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 680, in getMachineManager
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 1124, in createMachineManager
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 99, in __init__
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 349, in setActiveMachine
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Application.py", line 152, in setGlobalContainerStack
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212, in emit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\ExtrudersModel.py", line 140, in _extrudersChanged
AttributeError: 'NoneType' object has no attribute 'containersChanged'

-------------------------
Thread 0x00000e9c (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001fd4 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002648 (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001d68 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002540 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002a9c (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000015d0 (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002828 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000024d4 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000060c (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002a1c (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002914 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000021c4 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001b78 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000025a8 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001a18 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x00002100 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook

teriperi, 2017-08-17:
Hi @drdavelight  , I'm suspecting that this is the same issue as https://github.com/Ultimaker/Cura/issues/2243. Can you let me know what printers did you have to begin with before you upgraded to 2.7? 

Thanks!

Ghostkeeper, 2017-08-23:
No, this seems to be a duplicate of #2239.

closed 2017-08-23 by Ghostkeeper