Ultimaker/Cura #2286: 2.7 bug in machine settings dialog

sco01, 2017-08-17:
Tested with 2.7.0-Beta on OSX 10.12.6
Type: UI issue
Severity: Minor

There are a couple of usability issues and bugs with the new "tabbed" interface that depends on the number of configured extruders.  

1. When the number of extruders is set to 1:
The Start Gcode  and End Gcode editors in the bottom of the dialog only shows 5 lines of text at the time. This is really too small for an editor. Previous versions has shown at least 8 lines of code.
[image]

2. When the number of extruders is set to 2 or more:
a: There is a seven second delay before the dialog is updated to show the extruder 1 and extruder2 tabs at the top. The OSX Spinning Wait Cursor is shown during the delay.
b: The Start Gcode  and End Gcode editors in the bottom of the dialog now shows 7 lines of code since the nozzle size setting is moved to the nozzle tabs. This increased editing area for the gcode is better and should be the size when only one extruder is configured too. The Machine Settings dialog height could be increased slightly to improve the usability. 
[image]

3. The Gantry height defaults to "99999999999" mm for a newly created custom machine. (see screenshot above). That's a pretty big gantry for a 3d Printer :)

nallath, 2017-08-18:
Tagging @fieldOfView, as hes the one who made most of this in the first place

The gantry height is set to that so that it's not used.

larsarv, 2017-08-18:
> The Machine Settings dialog height could be increased slightly to improve the usability
On my 2013 13" Macbook pro retina screen this dialog is already to big. I can not see the lower part and have to put cursor in "end gcode" editor, press tab and then enter to close it...

maukcc, 2017-08-18:
Is there a way that when you drag the box bigger, it remembers the setting?

fieldOfView, 2017-08-18:
I can see that going all sorts of horribly wrong with combinations of low and high dpi screens and multiple monitors. I agree gcode editing can use some love (for example it would be nice if it had syntax highlighting and helped with {replacement_patterns}). Perhaps I can move the gcode fields to a tab of their own.

sco01, 2017-08-19:
I think moving the gcode fields to a separate tab is a good idea to begin with.  More available space is  better when editing code.

Syntax highlighting, helpers (code completion?) and validation/sanity check (if at all possible) are nice additions.

The gantry height issue is more cosmetic.

Ghostkeeper, 2017-08-29:
As for issue 3, the gantry height is set so high to simulate having no limit to the gantry. If the real gantry height is unknown, we thought it better to not limit the height of the volume when printing in one-at-a-time mode.

2a is a difficult problem. We might take another look at it in the fall because all of our tests currently indicate that the responsiveness of Cura front-end has dropped catastrophically again recently, when switching materials, variants, printers, etc.

For 1 I think we can afford to make the window a little bigger?

ianpaschal, 2018-03-05:
Slowness should be improved with the refactoring in 3.3. As for the UX issues, paging @LukeChen-Ultimaker.

nallath, 2019-07-16:
This has been resolved in the meantime.

closed 2019-07-16 by nallath