Ultimaker/Cura #2287: Feature Request: Show Z seam in layers view

sco01, 2017-08-17:
The Z seam placement is often crucial for the outcome of prints. Cura has several settings for the Z seam and it would be very helpful to have a way to visualise how the settings will affect the result prior to printing.

I propose a new checkbox in the View Mode: Layers dialog. When Color scheme "Line Type" is selected, show an additional check box labeled "Show Z seam". This setting should be associated with a color and the the start/end point (lets, say first and last 0.1 mm of travel for each layer) of the shell in the view should be painted in this color. A bonus of having the setting independent of the "Show Shell" setting is that only the seam placement could be visualized.

Some additional food for thought for the future: Wouldn't it be great if the user could "paint" the desired Z seam placement on the model in the view and feed this to the slicer as an addition to the  "User Specified" Z seam alignment setting. As a possible minimum: Show the bed X-Y coordinates in the view to provide the user with input on what to enter in the "Z seam X" and "Z seam Y" settings so the user doesn't have to guess, print and repeat.

Cheers!

smartavionics, 2017-08-17:
If enough people ask for a feature, it may get implemented by the Cura devs (or a contributor, like myself). Otherwise, it's up to individuals to do it (at least with open source, you have that possibility)

But is this really necessary? After all, look at this example layer view, the z-seam location(s) are clearly recognisable from the travel lines:

[image]

> As a possible minimum: Show the bed X-Y coordinates in the view to provide the user with input on what to enter in the "Z seam X" and "Z seam Y" settings so the user doesn't have to guess, print and repeat.

Cura 2.7 now supports a relative mode for the z-seam location that makes it easier to specify and produces more consistent results when the model is multiplied.

sco01, 2017-08-17:
I appreciate the feedback. Yes. It is easy to spot a seam when the travel lines are external. It is much harder when movements are internal to an object as the shell covers the travel lines:
[image]

The per-model seam placement in 2.7 is much appreciated. It gives consistency between parts. It does however not adress the issue of placing or observing the seam to begin with.

fieldOfView, 2017-08-17:
It could be interesting to have the option to draw the line types in a gradient of the main color (eg red to dark red for the outer wall) so you can see in which direction the layer is printed. This will also show things like seams.

sco01, 2017-08-17:
That's a really good idea. It removes the need for a separate setting completely.

smartavionics, 2017-08-17:
Yes, that sounds like it would be worth investigating - needs a UI bod to look into it.

Ghostkeeper, 2017-08-29:
Let's put that up as the feature request of this issue then: Change colours of layer view depending on progress within the layer.

Ghostkeeper, 2017-08-29:
Would it make sense to use a separate "Colour Scheme" entry for this, next to "Material Colour" and "Line Type"?

smartavionics, 2017-08-29:
While we're on the subject of layer view colour schemes, I would like to have an option to have colour based on velocity (like s3d and probably other slicers) give you.

Ghostkeeper, 2017-08-29:
Yeah that's something I wanted for a very long time as well. Sort of a different issue though.

Ghostkeeper, 2017-11-21:
The colour based on velocity has now been implemented and will be released with version 3.1. There's still no indication of the Z seam though other than hiding the shell and showing the travel moves bare. Our project manager rejected this and removed it from the planning for being more than 12 weeks old.

closed 2018-11-26 by Appesteijn