Ultimaker/Cura #2289: Cura 2.7 beta

stcs1981, 2017-08-17:
Cura 2.7 beta does not let me import the profiles of bq witbox 2

Ghostkeeper, 2017-08-29:
Where are you importing it from? Where did these profiles originate?

stcs1981, 2017-08-29:
I care since 2.5 cura and they appear to me when it is a superior cura do not appear, it is from the official website of bq.

stcs1981, 2017-08-29:
I care since 2.5 cura and they appear to me when it is a superior cura do not appear, it is from the official website of bq witbox 2

Ghostkeeper, 2017-09-03:
I can't find any profiles on their website. There is one thing listed [here](https://www.bq.com/en/support/witbox-2/support-sheet) under "Print profile", but that's just a printer definition made for Cura 15.10 2 years ago. It is now outdated and won't load in a modern Cura version. Thankfully, it won't need to, because modern Cura ships with a built-in profile for BQ Witbox 2 :smiley: We've kept that file up to date.

However I still can't find where you found any profiles that you could import in Cura.

closed 2018-02-27 by ianpaschal