Ultimaker/Cura #2290: Bug: Cannot display layers

kalsan, 2017-08-17:
This is a 64bit-based Arch Linux system with the following graphics configuration:

    OpenGL vendor string: X.Org
    OpenGL renderer string: Gallium 0.4 on AMD TURKS (DRM 2.50.0 / 4.12.6-1-ARCH, LLVM 4.0.1)
    OpenGL core profile version string: 3.3 (Core Profile) Mesa 17.1.6

Cura Version: `cura 2.6.0-2`

When clicking visibility -> layers, a bar appears, indicating that layers are being calculated, but the process never terminates.

Reproduce:

- Start cura
- Import some STL
- Select Material -> PLA [program lags for 1-2 seconds]
- View Mode -> Layers
- GUI says "Processing Layers 99%" forever

Console output at relevant part:

```
2017-08-17 22:42:57,997 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-08-17 22:42:58,676 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to LayerView
2017-08-17 22:42:58,958 - ERROR - UM.Logger.logException [76]: Exception: Job <CuraEngineBackend.ProcessSlicedLayersJob.ProcessSlicedLayersJob object at 0x7fcbc0a4dc88> caused an exception
2017-08-17 22:42:58,959 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-08-17 22:42:58,959 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3.6/site-packages/UM/JobQueue.py", line 125, in run
2017-08-17 22:42:58,959 - ERROR - UM.Logger.logException [80]:     job.run()
2017-08-17 22:42:58,959 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/CuraEngineBackend/ProcessSlicedLayersJob.py", line 178, in run
2017-08-17 22:42:58,960 - ERROR - UM.Logger.logException [80]:     color_code = material.getMetaDataEntry("color_code", default="#e0e000")
2017-08-17 22:42:58,960 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

```

kalsan, 2017-08-17:
The bug does not occur in Cura 2.6.2. I'll close this issue and let it here for future reference. Hope the Arch package will be updated soon.

kalsan, 2017-08-17:
Bug also occurs in cura 2.6.2.

It didn't occur on a fresh install, but when restoring .local/share/cura settings, the behavior re-appeared.

nallath, 2017-08-18:
I vaguely remember making some fixes here for 2.7. Could you check if the 2.7 beta still has the issue? 

The code does look different, so it's likely that my memory is correct.

kalsan, 2017-08-18:
Indeed, cannot reproduce in 2.7. Closing, thanks!

closed 2017-08-18 by kalsan

Ghostkeeper, 2017-08-29:
Yes, fixed here: https://github.com/Ultimaker/Cura/commit/6cac15de7d9b9eaf98fd8bef291ba4b6f0470791