Ultimaker/Cura #2291: Incorrect BQ profiles

aguspa, 2017-08-18:
Hi,

I have detected that BQ profiles are incorrect. Filament diameter should be 1.75mm. Besides, the profiles are not well adjusted (temperature, retraction...). If you want I can send you some good and tested profiles.

Thanks!

nallath, 2017-08-18:
By all means. The BQ profiles have not had updates in a long time. Ultimaker & Lulzbot are the only companies that are paying developers to create new features & update profiles, so the rest tends to be a bit outdated (unless community members fix it).
You can send the profiles to us, but it's preferred that you make a pull request for them. That way it's less work for us and we can always see who made certain changes.

aguspa, 2017-08-18:
I will make a pull request with all BQ printers profiles and filament corrected asap.

Thank you nallath.

closed 2017-08-18 by nallath