Ultimaker/Cura #2293: Master branch crash creating new printer (UM 3)

smartavionics, 2017-08-18:
I ask it to create a new printer and get this crash, happens repeatedly...

````
2017-08-18 10:19:28,461 - INFO - cura.Settings.GlobalStack.addExtruder [80]: Extruder[ultimaker3_extruder_left #2] added to [Ultimaker 3] at position [0]
2017-08-18 10:19:28,632 - INFO - cura.Settings.GlobalStack.addExtruder [80]: Extruder[ultimaker3_extruder_right #2] added to [Ultimaker 3] at position [1]
file:///usr/local/share/cura/resources/qml/WorkspaceSummaryDialog.qml:137: TypeError: Cannot read property of null
2017-08-18 10:19:31,302 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [392]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.024113178253173828 seconds. 121 input vertices. 121 output vertices.
Fatal Python error: Segmentation fault

Thread 0x00007efdd5232700 (most recent call first):
  File "/usr/local/lib/python3/dist-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdd5a33700 (most recent call first):
  File "/usr/local/lib/python3/dist-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdd77d3700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdf0b83700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdfe8b8700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdfe0b7700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007efe0ed67700 (most recent call first):
  File "/usr/local/lib/python3/dist-packages/UM/Qt/ListModel.py", line 63 in setItems
  File "/usr/local/lib/python3/dist-packages/cura/Settings/ExtrudersModel.py", line 224 in __updateExtruders
  File "/usr/local/lib/python3/dist-packages/cura/CuraApplication.py", line 683 in run
  File "/usr/local/bin/cura", line 73 in <module>
Segmentation fault (core dumped)
````

smartavionics, 2017-08-22:
It's still crashing when I try to create a new printer. What can I do to get more info about what's wrong?

nallath, 2017-08-22:
I don't know. None of us have had this issue. 

Are you running it from source or is this a release? If you are running from source, what library versions are you using?

smartavionics, 2017-08-22:
Hi. Which libraries are you referring to?

smartavionics, 2017-08-22:
OK, if I run gdb, I get the following when I create a new printer:
````
2017-08-22 14:24:36,949 - INFO - cura.Settings.GlobalStack.addExtruder [80]: Extruder[ultimaker3_extruder_left #2] added to [Ultimaker 3] at position [0]
2017-08-22 14:24:37,142 - INFO - cura.Settings.GlobalStack.addExtruder [80]: Extruder[ultimaker3_extruder_right #2] added to [Ultimaker 3] at position [1]
file:///usr/local/share/cura/resources/qml/WorkspaceSummaryDialog.qml:137: TypeError: Cannot read property of null
2017-08-22 14:24:39,912 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [392]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.023807048797607422 seconds. 121 input vertices. 121 output vertices.

Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007f3e0217e8fc in ?? () from /usr/lib/python3/dist-packages/PyQt5/QtCore.cpython-35m-x86_64-linux-gnu.so
````

smartavionics, 2017-08-22:
The PyQt5 version is 5.6+dfsg-1~cura~ppa~201609120729~6~ubuntu16.04.1

smartavionics, 2017-08-22:
I installed some debug stuff and gdb now reports:
````
#0  0x00007f800250c557 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007f7fff0b26bf in qt_metacall_worker (pySelf=pySelf@entry=0x7f7f9ca18a68, pytype=<optimised out>, base=base@entry=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=_c@entry=QMetaObject::InvokeMetaMethod, 
    _id=_id@entry=0, _a=_a@entry=0x99c11c0) at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:103
#2  0x00007f7fff0b25e2 in qt_metacall_worker (pySelf=pySelf@entry=0x7f7f9ca18a68, pytype=0x2d5b2e8, base=base@entry=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=_c@entry=QMetaObject::InvokeMetaMethod, 
    _id=_id@entry=0, _a=_a@entry=0x99c11c0) at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:84
#3  0x00007f7fff0b2a02 in qpycore_qobject_qt_metacall (pySelf=0x7f7f9ca18a68, base=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=QMetaObject::InvokeMetaMethod, _id=0, _a=0x99c11c0)
    at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:69
#4  0x00007f7ffec7bc47 in ?? () from /usr/lib/python3/dist-packages/PyQt5/QtQml.cpython-35m-x86_64-linux-gnu.so
#5  0x00007f800250cee0 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007f7fff0b26bf in qt_metacall_worker (pySelf=pySelf@entry=0x7f7f9ca18a68, pytype=<optimised out>, base=base@entry=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=_c@entry=QMetaObject::InvokeMetaMethod, 
    _id=_id@entry=0, _a=_a@entry=0x99c11c0) at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:103
#7  0x00007f7fff0b25e2 in qt_metacall_worker (pySelf=pySelf@entry=0x7f7f9ca18a68, pytype=0x2d5b2e8, base=base@entry=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=_c@entry=QMetaObject::InvokeMetaMethod, 
    _id=_id@entry=0, _a=_a@entry=0x99c11c0) at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:84
#8  0x00007f7fff0b2a02 in qpycore_qobject_qt_metacall (pySelf=0x7f7f9ca18a68, base=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=QMetaObject::InvokeMetaMethod, _id=0, _a=0x99c11c0)
    at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:69
#9  0x00007f7ffec7bc47 in ?? () from /usr/lib/python3/dist-packages/PyQt5/QtQml.cpython-35m-x86_64-linux-gnu.so
#10 0x00007f800250cee0 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007f7fff0b26bf in qt_metacall_worker (pySelf=pySelf@entry=0x7f7f9ca18a68, pytype=<optimised out>, base=base@entry=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=_c@entry=QMetaObject::InvokeMetaMethod, 
    _id=_id@entry=0, _a=_a@entry=0x99c11c0) at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:103
#12 0x00007f7fff0b25e2 in qt_metacall_worker (pySelf=pySelf@entry=0x7f7f9ca18a68, pytype=0x2d5b2e8, base=base@entry=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=_c@entry=QMetaObject::InvokeMetaMethod, 
    _id=_id@entry=0, _a=_a@entry=0x99c11c0) at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:84
#13 0x00007f7fff0b2a02 in qpycore_qobject_qt_metacall (pySelf=0x7f7f9ca18a68, base=0x7f7fff3773c0 <sipTypeDef_QtCore_QAbstractListModel>, _c=QMetaObject::InvokeMetaMethod, _id=0, _a=0x99c11c0)
    at ../../qpy/QtCore/qpycore_qobject_helpers.cpp:69
````
This repeats for over 10,000 function invocations so it looks like it is stuck in a recursive loop of some kind.

smartavionics, 2017-08-22:
I realise that the above library is probably too old so I'm currently building the environment from scratch using your repo and hopefully that will fix the problem.

nallath, 2017-08-22:
It's indeed an issue within the QT library that is causing the crash. So update pyqt & qt.

hdmonline, 2017-08-22:
I got the exactly the same problem with ubuntu16.04.1. Haven't figured it out yet.

smartavionics, 2017-08-22:
Hi, I am now building a new environment using https://github.com/Ultimaker/cura-build-environment. It's still building so I can't report whether that's the solution or not.

smartavionics, 2017-08-22:
I'm glad to say that using cura-build-environment appears to have fixed this problem. Closing...

closed 2017-08-22 by smartavionics