Ultimaker/Cura #2297: Ironing feature in 2.7.0 beta extrudes filament

bipsendk, 2017-08-19:
When trying to do ironing on a STL file, the target printer sounds like it it extruding a bit of filament some places on the layer being ironed.

It happens on the layer at 6.00mm (layer thickness 0.2mm).

Would you prefer to have both STL file and generated gcode file ?

fieldOfView, 2017-08-19:
That is by design. Extruding a tiny bit of material fills in tiny gaps while ironing. It has the additional benefit of keeping the nozzle primed while preventing material from degrading due to staying inside the hot hotend for too long.

bipsendk, 2017-08-19:
It just leaves some "bumps" on the object in question (places with small distance between mounting hole and edge of object) - not exactly what I was expecting. I think the ironing feature then probably is best suited for objects with larger flat surfaces without any holes or similar in them.

sco01, 2017-08-20:
The Ironing Flow setting (default: 10%) controls the amount of extrusion during ironing. You can try with a lower value and see if it helps.

nallath, 2017-08-21:
This is also why it's experimental; we haven't found perfect settings for it yet. Finding those is left "As an exercise to the reader" for the moment being. We do want to hear what values you found to work though.

closed 2017-08-21 by nallath