Ultimaker/Cura #2299: Cura 2.7 Beta Python 3.5 calls nonexisting "things"

dintid, 2017-08-20:
Cura 2.7 beta has started making calls for drive D... I only have drive C. 
```
Version: 2.7.0-BETA
Platform: Windows-10-10.0.15063
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 772, in setActiveMaterial
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 835, in setActiveQuality
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 952, in _replaceQualityOrQualityChangesInStack
AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

-------------------------
Thread 0x00001f60 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001dbc (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000042a4 (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00003158 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00003a6c (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000047b8 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00003d00 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00003f24 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
[... 74 lines cut ...]
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001cc8 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x00000158 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 676 in run
  File "D:/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
```

nallath, 2017-08-21:
Well, you do. It's not making calls to the file, it's telling that it's using a file that is on the D drive.

fieldOfView, 2017-08-21:
No, it is not. It is an artifact of the build system. The files are on the D drive on the build system, not on the D drive on the user's system.

dintid, 2017-08-22:
It's rather sad that you focused on telling me I'm wrong in how I formulated the bugreport instead of talking the problem seriously.

So, what to do about this problem? I noticed there is a 250mb old log file and LOTS of old files in Cura folder/s... any way to make Cura run smoother and not take 57,9 seconds to load? It takes ages on machines for one...

fieldOfView, 2017-08-22:
I was responding to @nallath

fieldOfView, 2017-08-22:
@dintid, from what I know, your problem is not caused by Cura "referencing" files on a non-existent D-drive. It has been doing so since Cura 2.1, you just never noticed it before.

The relevant bit of the first couple of lines of the crash report are ```cura\Settings\MachineManager.py", line 952, in _replaceQualityOrQualityChangesInStack
AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'```
This points to an unexpected event, regardless of the D: drive that was referenced. It is probably caused by an incorrect upgrade action performed by Cura on the configuration of Cura 2.5 or Cura 2.6 to Cura 2.7.

Thisismydigitalself, 2017-08-22:
https://github.com/Ultimaker/Cura/issues/2261

Ghostkeeper, 2017-08-23:
Indeed the cause of this is that these Python files are frozen on our build server, then packed into an executable for Windows. On our build server these files are located in `D:\2.7\build\...` somewhere and cx_freeze takes that path as the value for the `__file__` attribute of the Python module. That's what then ends up in the log. The actual files don't exist any more since they are packed inside an .exe file.

This isn't really a bug or problem, but a side effect of how we package our application. The log is perfectly usable as long as the last few bits of the path within the Cura directory are correct.

closed 2017-08-23 by Ghostkeeper