Ultimaker/Cura #2301: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance

jerry7171, 2017-08-20:
_Probably the same error as issue #1932. I was just preparing to close Cura after slicing an E3D v6 mount for the Monoprice Select Mini v1. I succeeded in saving the gcode when this error occurred._

Version: 2.7.0-BETA
Platform: Darwin-15.6.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2
 
Exception:
TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance
 
 
-------------------------
Thread 0x000070000757d000 (most recent call first):
  File "/Applications/Cura 2.7.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
 
Thread 0x000070000707a000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
 
Thread 0x0000700006b77000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
 
Thread 0x0000700006674000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
 
Thread 0x0000700006171000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
 
Thread 0x0000700005c6e000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
 
Thread 0x000070000576b000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
 
Thread 0x0000700005268000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
[... 113 lines cut ...]
  File "/Applications/Cura 2.7.app/Contents/MacOS/plugins/plugins/Tools/CameraTool/CameraTool.py", line 134 in initiateZoom
  File "/Applications/Cura 2.7.app/Contents/MacOS/plugins/plugins/Tools/CameraTool/CameraTool.py", line 150 in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Controller.py", line 247 in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315 in __performEmit
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 212 in emit
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtMouseDevice.py", line 40 in handleEvent
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Qt/Bindings/MainWindow.py", line 158 in wheelEvent
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 676 in run
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
 
[image]

Ghostkeeper, 2017-08-29:
I suspect it's this one:

https://github.com/Ultimaker/Cura/blob/2.7/cura/CuraApplication.py#L850

However without reproducing this myself I can't tell for sure and can't debug the cause. Something that should be boolean contains None somewhere in something that listens for that event, I suspect.

ianpaschal, 2018-03-09:
This hasn't been touched in 6+ months so I'm closing it.

closed 2018-03-09 by ianpaschal