Ultimaker/Cura #2302: Multiplying Models compromises internal geometry

billix0, 2017-08-21:
I'm experiencing an issue where using the "Multiply Selected Models" function is causing the object's internal geometry to be erased and just replaced with infill. It's behaving as if the Mesh Fixes' Remove All Holes option is selected when it isn't. Cura 2.6.1.

fieldOfView, 2017-08-21:
Duplicate of https://github.com/Ultimaker/Cura/issues/1077
TL;DR: your model is broken. With only one model on the buildplate, CuraEngine just happens to be able to figure out what is going on. With multiple objects, it fails to do so.

Try slicing with these two settings from the Mesh Fixes category:

Union overlapping values: unchecked
Merged meshes overlap: 0

You can use the "Search" box on the top of the custom settings to search for these settings.

nallath, 2017-08-21:
Multiply selected models indeed creates exact copies of the model. I very much doubt that this is the issue. But just to be on the safe side, what happens if you load the same model 3-4 times. If it's the same, it's not the multiply model that's the issue.

ianpaschal, 2018-03-02:
Needs info but never got it. Closing.

closed 2018-03-02 by ianpaschal