Ultimaker/Cura #2303: Cura loading _extremely_ slowly, while you say it's faster now?

dintid, 2017-08-21:
I don't know if this can be labeled a bug or anything, but it's been so bad I just _have_ to know now.

Cura starts up slower than large games with super many textures, models etc like Civ VI and similar SIM games.
Or to compare to same category, my Fusion 360 starts up faster than Cura.. that can't be right?

My PC is in perfect order hardware and windows 10 wise, but I know next to nothing about Python, which I suspect is the culprit... lately it made a call to a D drive I don't have!

**Hardware**:
i7 3770K
16gb ram
SSD drives
Nvidia GTX 1050 Ti

nallath, 2017-08-21:
Cura boots in about 13-15 seconds with me. This isn't really fast considering that with C++ you can probably get it down to seconds (or less). We still have some legacy things that work but are slow. We are changing those,  but don't want to change everything in one go (as this will probably break things). The speedup time used to be worse (40-45 seconds), so pretty big improvements were made already. 

As for the call to the D drive, this has been fixed for 2.7

Ghostkeeper, 2017-08-30:
It did become slower between 2.6 and 2.7 for me, actually. Or at least, it feels that way.

In any case, this is a known issue. I suspect that we'll get some time for speed improvements again in the fall because I don't see that happening for the release after 2.7 either...

Thisismydigitalself, 2017-09-13:
Loading machines takes very long time. does machines = printers? if so, i don't need them all. only my printer. where to located all relevant json files which can then be moved aside, thus making 2.7 load faster?
Generally speaking - what should one practice in order for Cura to load faster?

nallath, 2017-09-14:
Removing materials that you don't need, removing definition files that you don't need, profiles, etc All of those will have an impact.

ChrisTerBeke, 2018-01-22:
We're bringing speed increases in 3.2 and continue to work on this in the future. Closing for now. Please re-open if 3.2 is still unusable for you.

closed 2018-01-22 by ChrisTerBeke

Thisismydigitalself, 2018-01-23:
My benchmark for 3.1:
32 seconds loading time on my i5 with 32Gb RAM running on Win10/64.

fieldOfView, 2018-01-23:
A 3.2 beta is available now. Could you test that?

davemcguire, 2018-03-02:
3.2.1 (via appimage) sits at "loading machines", five minutes and counting.  Platform is current Linux Mint, i7, 24GB of RAM.

Ghostkeeper, 2018-03-28:
Wow, sounds serious. How big is your folder `~/.config/cura`? And is there anything in the log file?

davemcguire, 2018-03-28:
I figured out what was going on, but only after half of the equipment in my lab had gone haywire.

In one of the most astonishingly foolish development decisions I've seen in my career, the Cura developers apparently thought it was a great idea for Cura to squirt crap out of EVERY SERIAL PORT IT FINDS looking for a 3D printer.  It's difficult to imagine a more egregiously bad thing for an application to do.  I am an embedded systems developer; at any one time I've got five or six /dev entries on my system that represent USB-serial bridges.  Some of those go to prototype boards that I'm working on for clients, and some go to electronic test equipment on the test bench.

I was eventually able to work around it, and Cura starts up normally when it can no longer open those serial ports.  But wow, guys.  There are serial things that are not printers, and maybe even printers that we don't want Cura to automatically assume it should be messing with!  It's an awful idea to just randomly open up serial ports and start squirting crap out at them when you don't have any clue of what they might be.  That "feature" needs to be removed.

                        -Dave

fieldOfView, 2018-03-28:
Go to the Plugins pane of the preferences and untick the USB Printing box. Cura will no longer probe all USB ports (but also won't be able to print over USB anymore).

davemcguire, 2018-03-28:
Thank you for your suggestion.  I cannot open the preferences dialog if Cura is hung trying to start up, but this is a good approach once it's up and running via some other method.  As it was, for me, Cura was sitting apparently locked up for twelve minutes (all the while squirting crap out of my serial ports!) before I finally killed it.

If I were a robotics engineer, I could conceivably have been physically injured by this program's behavior.

If the Cura developers feel that they really MUST cater to the masses who refuse to learn how to determine their printer's /dev entry, then this scan function should be in the printer setup dialog, with a user-initiated "scan all serial ports looking for a printer" option.  This would take about ten minutes to code up (yes, I'm sure lots of people say things like that, but I'm a programmer too...that really is a ten-minute job) and would be far safer and make Cura a well-behaved application with regard to other devices.

                             -Dave

fieldOfView, 2018-03-28:
Hey, I am just an external contributor, trying to be helpful. I get that you are frustrated, but try to stay productive.

I can assure you that what you suggest is not a 10 minutes job in Cura. Why? Because I tried it before. For one thing, Cura does not have a “printing dialog” to move the scan to. It has a “monitor” that expects a connected printer. I have tried to overhaul the port opening behavior before, but other things got in the way. There’s only so much time I can contribute in a week. I do mean to revisit that work, possibly for Cura 3.4.

If you were a robotics engineer, you would have known better than to control a potentially life threatening robot from the same computer as your 3d printer. But I get your point, and agree with it.

If you are comfortable editing a configuration file in a texteditor, I can tell you how to disable the usb printing plugin without starting Cura.

davemcguire, 2018-03-29:
You are right Aldo, I apologize.  I did look in the config file and wasn't able to spot anything about USB printing; perhaps I missed it.  I'm (also) a long-time commercial UNIX admin, so I'm fine with editing text files.  While I've already addressed the problem another way, if you could tell me where that is in the config file, I'd appreciate it, just so I'll know for the future.

About the printer setup dialog...I haven't looked at it in a week or two, but I could swear there was a printer configuration dialog box in which I added new printers.

Thank you for your reply, and for your efforts.

             -Dave

fieldOfView, 2018-03-29:
Think nothing of it. Frustration happens.

I had to look up the exact syntax for the fix in the configuration file. Add the following line to the [general] section of ~/.config/cura/3.2/cura.cfg (while Cura is not running, but that should not be a problem for you):
```disabled_plugins = USBPrinting```
This will stop Cura from trying to access your serial ports.

ktims, 2018-04-11:
Having exactly the same issue, hanging at 'Loading machines...' with no console output and no indication of progress.

I agree with @davemcguire , if his assessment of the serial port behaviour is correct, that's absolutely unacceptable. It's even more frustrating that the program can't even start with $not_3d_printer attached to a serial port.

Anyway, what can I do to start Cura? I'd really rather not go spelunking in udev configuration and messing with permissions to block Cura from messing with the ports, and adding ` disabled_plugins = USBPrinting` in the config file (which does not exist, though it does create `.config/cura`) doesn't seem to do anything.

davemcguire, 2018-04-11:
I ended up spelunking in my udev configuration. ;)  Since I'd already had it handled that way, I didn't try the config file modification, but I'm surprised to hear that it didn't work.