Ultimaker/Cura #2306: master branch - OTT spaghetti infill

smartavionics, 2017-08-22:
Just tried enabling the spag infill, didn't alter any params, came out rather thick!

[image]

nallath, 2017-08-23:
That's why it's experimental ;)

smartavionics, 2017-08-23:
Indeed, do you want any more info re settings, system, etc.?

nallath, 2017-08-23:
@BagelOrb; Is this something we want to improve on?

BagelOrb, 2017-08-23:
Errrr. I don't really understand what's expected from me here.

Spaghetti infill works weird. It is generating lines which are supposed to fill downward, but the frontend is interpreting those extrusion paths as being very thick. The frontend cannot simulate how the filament will curl up into spaghetti.

Ghostkeeper, 2017-08-31:
I think we might have to reduce the default setting for maximum height to 1.0mm instead of 2.0. Perhaps that works better in most printers.

BagelOrb, 2017-08-31:
I haven't tested that much. That's why it's experimental. I have no idea whether 1mm works better than 2mm. Did you test this @Ghostkeeper ?

Ghostkeeper, 2017-09-10:
No, I didn't test that. It was just a feeling. 1mm is plenty to create spaghetti, but you might get more reliable extrusion. I did test the default settings for spaghetti infill and that just produced a giant blob of plastic on one of the walls.

Even with 1mm layers, you'll still get humongous lines. That's just the way that the layer view visualises it, not knowing that the material will/should just fall down.

ianpaschal, 2018-03-05:
Closing because this is 6+ months without activity. Exper'mental features gon' exper'ment.

closed 2018-03-05 by ianpaschal