Ultimaker/Cura #2307: Aborting print

Jhinta, 2017-08-22:
When printing with 2.6 and pressing abort, the print just wil halt
When printing with 2.7 and pressing abort it seems like its restarting start-code. ( home y-x and z)

Both 2.7 and 2.6 use the same start and end code.
Whats wrong here?

End-Code
`M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
G90 ;absolute positioning
G1 Y190 F9000 ;use this line if you want the bed to move to the front. delete if not.
M84 ;steppers off`

Ghostkeeper, 2017-08-31:
Yes, I added this here: 6d2b46244f910cc5f09b07bdde1955bdcde77afb

Otherwise when you abort, it would keep the nozzle stuck on the print and make a mess. This was intended.

It's not the start or end code that it executes. It just homes the nozzle now.

Ghostkeeper, 2017-08-31:
Is there anything wrong with this approach?

Jhinta, 2017-08-31:
Yes, its also homes Z , it sould not do that .
only do home all but Z (Z save zone problems !)
And why would one do a hame z when there is a abort?

Jhinta, 2017-08-31:
Yes, its also homes Z , it sould not do that .
only do home all but Z (Z save zone problems, and when you have a big/multi project on border it will distroy your project) !)
And why would one do a home Z when there is a abort?

Jhinta, 2017-08-31:
more like this i think , move Z up a litle and then home Y & X
self._sendCommand("G1 Z+0.5 ")
self._sendCommand("G28 X Y")

nickthetait, 2017-09-01:
@Ghostkeeper this is a dangerous operation only for printers that home Z at the bottom

fieldOfView, 2017-09-06:
It is also an issue with Ultimaker Original, which homes at the bottom. I would say it is not necessary to home the z-axis at all.

A longer term solution would be to have an abort-gcode machine-setting in Cura (possibly along with pause-gcode and resume-gcode settings).

Ghostkeeper, 2017-09-10:
Thanks to @fieldOfView's fix, it now only homes X and Y, not Z.

closed 2017-09-10 by Ghostkeeper

Ghostkeeper, 2017-09-10:
Now that I think about it, the same can happen when printing in one-at-a-time mode, when aborting while printing the second object.

fieldOfView, 2017-09-10:
Moving up 10mm in relative mode may be all that can be done safely.

jack0088, 2017-10-13:
So is this fix included with 2.7.0 or when will it be? Because on my Prusa i3 MK2 it just home'ed everything on canceling (including the Z-axis and crashed into the printing).

fieldOfView, 2017-10-13:
This is fixed with Cura 3. A beta is available through the Ultimaker forum

EnumC, 2017-11-09:
@fieldOfView Running Cura 3.0.4 and it's still auto homing Z on abort. This is especially dangerous for machines that use fixed probe, as auto homing Z will cause the probe to go out of bed and crack the glass bed underneath.

fieldOfView, 2017-11-09:
Unfortunately a number of fixes were made to the master branch before the release of 3.0 that did not get included in 3.0. This is one of them. This is probably not going to be fixed until Cura 3.1.

asantoni, 2018-02-27:
This is still an issue as of Cura 3.2.0. Can we reopen this issue to reflect the bug status?

collse, 2018-11-07:
has this been addressed? Where can I change the abort gcode in Cura 3.5.1?

SteveC00k, 2019-01-12:
When the printer homes on my printer because of the sides it hits the edge and bends the hotend. Why would you not place the ability to control this. :/

PR #2391 (2017-09-10): Remove home bed from abort sequence
This PR removes homing the bed from the abort sequence. On printers that home the bed to the bottom (ie: printhead moves towards the bed), homing the bed when aborting a print could cause the printhead to crash into the aborted print.

Fixes #2307 and #2344