Ultimaker/Cura #2311: Tesalating objects won't connect

OneLooseCrank, 2017-08-23:
Hi all,
I've had problems in the past making parts that will connect together. I assumed that Cura would create a line that follows the perimeter of the object, meaning that half of the printed line will fall outside of the object's designed perimeter. I've recently been informed that this is not the case and that the slicer should 'shrink' the printed object to compensate for the diameter of the printed line, and thus never create an object bigger than it was designed (excluding tolerances). Can anyone confirm which is true, and if it is that Cura automatically compensates for printing thickness, any thoughts on why I have been unable to print male and female parts that properly intersect?

Cheers

nallath, 2017-08-24:
Cura can compensate for shrinking (and create an object that is bigger than designed). The horizontall offset setting is used for this. 

If you want to have perfectly fitting connectors, you will need to change tolerances based on the material, as every material behaves slightly different

DDDirk, 2017-08-24:
Also, prints are hardly ever completely perfect.
For example, the filament (especially the cheaper kind) may not have a constant width, resulting in under- or overextrusion. Or if (part of) an object is very small the layers may not have enough time to cool down and droop a little. Tweaking can help, but is it difficult to make a slicer that accounts for every possible geometry. Oozing can also create problems (also depending on the material). And there are many more factors.

You could print the critical parts first, to check which settings work best.
Another option is to make holes and/or pins slightly conical instead of straight. And/or curved (eg a 'hollowed out' hole), so that only the base and end of an inserted pin are supported (where it matters). Those are just 2 ideas. 3D printing gives many more possibilities for shapes than substractive manufaturing (you couldn't easily make such a 'hollowed out' hole with a normal drill bit). Designers have to completely rethink how they design things, and that may take decades. In the meantime, be creative. :)

OneLooseCrank, 2017-08-26:
Thanks both,
Can you confirm if what I've been told is correct - that the slicer should not be putting down material outside of the boundary of the object though as I'm still not clear on this. Essentially, if I've got a .4mm nozzle, is it tracking 0.2mm away from the boundary assuming that with nozzle width will lay down .2mm either side of its path and thus print up to the edge of the object but not beyond it (excluding expansion / contraction / tollerances).
I see your point about contraction of the material on cooling. I've started to print some test connections to see if I can design them to be a really close fit, but Cura has just upgraded to 2.6 and I'm having some issues with it atm... I'll be running some test parts off to see if I can get greater precision in size measuring with calipers after each attempt.

Ghostkeeper, 2017-09-02:
Indeed, Cura moves the perimeter inside by half the (outer wall) line width in order to make the outside of the object line up exactly with the perimeter of the model.

closed 2018-03-05 by ianpaschal