Ultimaker/Cura #2313: fatal error !

Tartempion3, 2017-08-23:
Version: 2.7.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "D:/2.7/build/inst/bin/cura_app.py", line 72, in <module>
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Application.py", line 316, in getInstance
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 205, in __init__
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 110, in __init__
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 509, in _loadPlugins
RuntimeError: Could not load the backend plugin!

-------------------------
Thread 0x000004ec (most recent call first):
  File "C:\Program Files\Cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001920 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000112c (most recent call first):
  File "D:\env\2.7\inst\lib\socket.py", line 575 in readinto
  File "D:\env\2.7\inst\lib\http\client.py", line 258 in _read_status
  File "D:\env\2.7\inst\lib\http\client.py", line 297 in begin
  File "D:\env\2.7\inst\lib\http\client.py", line 1197 in getresponse
  File "D:\env\2.7\inst\lib\urllib\request.py", line 1257 in do_open
  File "D:\env\2.7\inst\lib\urllib\request.py", line 1282 in http_open
  File "D:\env\2.7\inst\lib\urllib\request.py", line 444 in _call_chain
  File "D:\env\2.7\inst\lib\urllib\request.py", line 484 in _open
  File "D:\env\2.7\inst\lib\urllib\request.py", line 466 in open
  File "D:\env\2.7\inst\lib\urllib\request.py", line 163 in urlopen
  File "C:\Program Files\Cura\plugins\UpdateChecker\UpdateCheckerJob.py", line 47 in run
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 125 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001128 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00000df4 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000073c (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000012a4 (most recent call first):
  File "C:\Program Files\Cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000011f4 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000002a4 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00000fa4 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001a58 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00000f24 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x00001194 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook

nallath, 2017-08-24:
Could you also post the cura.logs file?

Tartempion3, 2017-08-25:
Ok, i've made some manipulations ...
i have uninstalled and scratch all residual files for make a new install ...
the error is still here !

[cura.log.txt](https://github.com/Ultimaker/Cura/files/1252626/cura.log.txt)

Tartempion3, 2017-08-30:
Sorry for the "close issue" it's a mistake that i never want. 
Sorry for my english ... i'm a frenchie ;-)

Ghostkeeper, 2017-09-03:
Hmm, the actual error seems to occur before the file logger was even loaded.

Tartempion3, 2017-09-03:
Hi !
May be, i didn't read the log file.
That i know and don't understand is that i used several time cura 2.7 without any problem and "i'm fall in love with it" !
I didn't change anything on my config, didn't install soft, nothing.
But a day ... BOUM ... the fatal error occurs. Why ? Don't know !
I tried to uninstall and scrath all directories and files (i think) then i 
Reinstall the appllication as a first time  but the error persist.
I like to say "what the fuck" ! I'm so sad !
Sorry for my english writing ... i'm always a frenchy !
Best regards.

Ghostkeeper, 2017-09-10:
Deleting the folders at `C:\Users\<Tartempion3>\AppData\Roaming\cura` and `C:\Users\<Tartempion3>\AppData\Local\cura` will reset Cura to factory defaults.

ianpaschal, 2018-02-27:
@Tartempion3 Were you able to get this working?

closed 2018-03-02 by ianpaschal