Ultimaker/Cura #2315: Cura 2.7: Editing new material deletes all other materials

kaimikael, 2017-08-24:
As shown in the attached video, I duplicated material and when editing the properties of the new material, all other materials were deleted.  I can get around this by deleting the ~/Library/Application Support/cura/2.7/ directory and rerun Cura, which will restore the materials.
I also attach the cura log file.
I run Cura 2.7 beta under OS X 10.10.5.
[cura.zip](https://github.com/Ultimaker/Cura/files/1248243/cura.zip)
[video.zip](https://github.com/Ultimaker/Cura/files/1248248/video.zip)

nallath, 2017-08-24:
This should be fixed for 2.7 final release. I just tested this myself and it did not happen for me anymore. Could you retest this when the final is out?

kaimikael, 2017-09-04:
I can confirm that in 2.7.0 the materials are no longer deleted when editing a duplicate.

closed 2017-09-04 by kaimikael