Ultimaker/Cura #2323: cura 2.6.2

JAUYEB, 2017-08-26:
Spanish traslation is not correct:

"nozzle" not is "tobera" in spanish, is "boquilla".

Thanks a lot!

Ghostkeeper, 2017-09-03:
Wikipedia named it `tobera`, but both seem to be correct according to my dictionary. `boquilla` seems to indicate more of a beak-type thing than an orfrice-type thing.

I can't change this myself as I am no native speaker of Spanish. Changing the wording may also change the inflections and particles surrounding the word `tobera`, so a native speaker of Spanish will have to make these changes in the Spanish .po files.

JAUYEB, 2017-09-03:
I'm a native speaker of Spain.  "tobera" is big, "boquilla" is small. The nozzle is small.

I will change the .po files. 

Thanks!

closed 2018-02-27 by ianpaschal