Ultimaker/Cura #2324: Cura 2.7 Crash Report

dukemagus, 2017-08-26:
As directed:

Version: 2.7.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\Bindings\MainWindow.py", line 131, in mouseMoveEvent
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtMouseDevice.py", line 30, in handleEvent
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212, in emit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Controller.py", line 261, in event
  File "C:\Program Files\Cura 2.7\plugins\Tools\RotateTool\RotateTool.py", line 117, in event
    drag_start = (self.getDragStart() - handle_position).normalized()
TypeError: unsupported operand type(s) for -: 'NoneType' and 'Vector'

-------------------------
Thread 0x00001f70 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000006bc (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001778 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00000b1c (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001cec (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x00001c44 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 676 in run
  File "D:/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

JAUYEB, 2017-08-26:
me too, is the same issue?

-------------------------------------------
Version: 2.7.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 707, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189, in call
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderStack.py", line 94, in _onPropertiesChanged
AttributeError: 'NoneType' object has no attribute 'definition'

-------------------------
Thread 0x00002e2c (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001634 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000263c (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002990 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000016c4 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001288 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00000588 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
[... 76 lines cut ...]
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00000fa4 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x000029c0 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 676 in run
  File "D:/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

dukemagus, 2017-08-26:
> File "", line 634 in _load_backward_compatible
> File "", line 664 in _load_unlocked
> File "", line 958 in _find_and_load_unlocked
> File "", line 969 in
> File "", line 969 in _find_and_load

No idea. i reported so anyone that actually works on the project could be
notified.

in my case, it crashed while rotating a part

Thisismydigitalself, 2017-08-26:
I had the same issue. to my question asking for a workaround i got no answer.
https://github.com/Ultimaker/Cura/issues/2261
couple of days ago i downloaded cura-master and copied all files to cura 2.7 folder (running over existing files) and voila! - 2.7 is working for me just fine and looking great i might add.

This is by no means recommendation for you to do the same as i don't want to be blamed for advising something which can cause you problems - i'm only saying what worked for.

fieldOfView, 2017-08-26:
@JAUYEB and @Thisismydigitalself, your crash is a different issue than the one @dukemagus has reported. 
It helps to tell what you were doing when the crash happened, so it is easier for other people to see if their issue is similar.

Thisismydigitalself, 2017-08-26:
Nothing special. 
downloaded  and installed 2.7 BETA alongside 2.6.2 on win10/64.
2.6.2 worked (still do), 2.7B did not. replacing Cura 2.7's folder's files with Cura-Master files fixed it.

Ghostkeeper, 2017-09-02:
I've seen this issue come up from time to time, but it is never reproducible. So I think I'll now sweep it under the rug. See this commit: https://github.com/Ultimaker/Cura/blob/24c9dc9bee91cc505a819218288deeff3fe03953/cura/Settings/ExtruderManager.py#L434

Hopefully it won't occur again. Perhaps there is some multi-threading involved that sets the rotation plane back to None or something.

ianpaschal, 2018-02-27:
Since this hasn't been updated in 6 months and is not reproducible, this issue will be considered closed.

closed 2018-02-27 by ianpaschal