Ultimaker/Cura #2325: Rearranged printjob area

fieldOfView, 2017-08-26:
Recently, part of the printjob area was moved into the sidebar. When watching the print monitor, the data that is now shown in the sidebar suggests it is relevant for the currently monitored print, which may not be true.

**Steps to reproduce:**
* Load and slice an object
* Send it to the printer for printing (either through USB, UM3 or OctoPrint)
* Go back to the settings tab and load another model, change settings etc.
Note that the time in the sidebar will reflect the changes
* Now switch back to the monitor tab to check in on the ongoing print
Note that the time in the sidebar is now confusinglt not relevant to the currently printing job

The previous design purposefully subdued the viewport and the print job stats area to prevent this confusion.

An additional problem is visible here:
[image]

nallath, 2017-08-28:
I've notified our UX guys.

Ghostkeeper, 2017-09-03:
Did you get a response yet, @nallath?

nallath, 2017-09-04:
Nope. But i'm asuming any next re-design will take this into account. 
I did hide the print time widget in monitor as it was confusing.

ChrisTerBeke, 2017-11-14:
Recent UI changes fixed this (?)

closed 2017-11-14 by ChrisTerBeke