Ultimaker/Cura #2326: blank window/pulldowns immediately after creator's update in win10

punkonjunk, 2017-08-27:
Been chasing this around all night. I have multiple variously sized displays with various resolutions and various different DPI settings. Primary display is 4k display at 200% scaling.

I have two graphics cards, one driving the primary display, the other driving the secondary displays. It will behave as in one of the two pictures, regardless of which display it is launched on, and regardless of which display it is dragged to. It will either look normal and sort of work, but NO pulldowns work. Or, pulldowns will exist, but all be blank, in addition to the application being entirely blank.
[image]
[image]

I tried setting all displays to 1080p and 100% dpi, and having only one display, and I deleted all cura folders in appdata and roaming appdata and programfiles and such. I tried complete uninstall and reinstall, and I tried so so many things I could think of. I spent several hours tinkering with the compat settings for the shortcut, including the now weird ignore DPI settings, as admin, in win 7/vista/8 compat modes, as admin, etc. I have run out of ideas to try.

[logs.zip](https://github.com/Ultimaker/Cura/files/1254146/logs.zip)

I have attached my DXdiag to provide more information about my admittedly somewhat crazy setup, in addition to all cura logs I could find. Let me know if there is any more information I can gather for this issue, although I suspect the solution might be "your computer is crazy and it's your fault" although it is worth noting this worked totally normally before the creator update.  I am a competent operator and am willing to try any offered suggestions, rather than roll back.

punkonjunk, 2017-08-27:
I rolled back to pre-creators. It still does it on some displays, but works normally on my 4k display when I check the ignore DPI scaling in compat settings for the shortcut. 

I heavily beleive this issue is related to DPI, and my best guess to replicate it is a couple displays with different resolutions and DPIs. I am gonna mess with it on my laptop and see if I can develop a replication method for this bug next week. I'd really like to go up to the creators update as it has some very cool things I want to touch. (Baked in nix terminal!)

nallath, 2017-08-28:
We've noticed that certain multi monitor setups don't play well with QT (or qt doesn't play well with them). As we heavily rely on qt for a slew of things, there are certain issues that are outside of our power to influence.

Ghostkeeper, 2018-03-02:
Link for reference: https://github.com/GoldenCheetah/GoldenCheetah/wiki/FAQ-MOST-COMMON-QUESTIONS-OF-ALL#how-do-i-fix-tiny-text-on-high-dpi-displays

Ghostkeeper, 2018-03-02:
I think that flag won't work via PyQt though.

diegopradogesto, 2019-09-10:
Hi @punkonjunk , last time you commented on this thread was a long time ago. Are you still using the 2.x version of Cura? I encourage you to use a newer version and see if this issue is still not working for you, we can investigate. Otherwise I think we could close this issue.

mahtDFR, 2020-02-19:
There has been no activity on this issue for 30 days after information was requested, so I'm closing it.

closed 2020-02-19 by mahtDFR