Ultimaker/Cura #2328: Cura 2.6.2 - "Minimum Layer Time" causes slicing error

llamasoft, 2017-08-27:
**Application Version:** Cura 2.6.2
**Platform:** Windows 10 x64
**Qt:** N/A
**PyQt:** N/A
**Display Driver:** N/A

**Steps to Reproduce:**
Use default Prusia i3 machine profile with default PLA as material profile.
Set "Cooling -> Minimum Layer Time" to a value >= 11 seconds.

**Actual Results:**
> Unable to slice with the current settings.
The following settings have errors:
Prime Tower Y Position
Regular/Maximum Fan Speed Threshold

**Expected results:**
I expect it to slice normally, but simply run at slower speeds on small layers.  
If there was a slicing error, I also wouldn't expect the "Prime Tower Y Position" to be dependent on the "Minimum Layer Time", especially when the default Prusia i3 machine profile A) only has one extruder and B) the "Prime Tower" feature is disabled.

**Additional Information:**
If the "Minimum Layer Time" is greater than or equal to the "Regular/Maximum Fan Speed Threshold", it should default to regular fan speed for the whole print, right?  Every layer is going to take longer than the threshold, so every layer should use the regular, not the maximum fan speed.

nallath, 2017-08-28:
Does this also occur for 2.7?

llamasoft, 2017-08-28:
Cura 2.7 Beta kind of fixes it.  It no longer reports a slicing error for "Prime Tower Y Position", but it still reports "Regular/Maximum Fan Speed Threshold".

They both seem to be okay with "Minimum Layer Time" == "Regular/Maximum Fan Speed Threshold", but "Minimum Layer Time" > "Regular/Maximum Fan Speed Threshold" results in a slicing error.  
Shouldn't the result be running at regular fan speed for the whole print because each layer will take longer than the threshold?

Ghostkeeper, 2017-09-03:
Indeed this is still a bug in 2.7 and master.

Or rather, it was actually designed that way because we weren't thinking outside of the box.

I agree that we should fix this by allowing the Minimum Layer Time to be longer than Regular/Maximum Fan Speed Threshold but then maintaining the regular fan speed.

Ghostkeeper, 2017-12-04:
I made a small fix that addresses this issue: https://github.com/Ultimaker/CuraEngine/commit/990d1ae4190b0076594c317d6e5a62bec2a0e5f1

commit b453e0a4 (2017-12-04): Allow minimum layer time to be higher than the fan speed min/max threshold

Previously that wasn't possible because the division that calculated the slope of the fan speed ramping up towards the minimum layer time couldn't deal with negative values. Now it deals with that properly and in the limit the regular fan speed will be used.

Fixes #2328 and CURA-4272.
closed 2017-12-04 by Ghostkeeper