Ultimaker/Cura #2332: Lock W10 disable current timer

C4DG33K, 2017-08-28:
When locking and unlocking your W10 PC the timer is disabled and only possible to restart by closing and restarting Cura 2.6.2 again.
Windows: 10 Pro Eng 64-bit Version 1703 Build 15063.540
Cura: 2.6.2

Ghostkeeper, 2017-09-03:
Which timer do you mean? The remaining print time when printing via USB cable? Or the pre-heat timers?

C4DG33K, 2017-09-06:
The timer counting down the remaining print time.
Uses a Ultimaker 3 Extended via wired network, not wifi.

I have not had time to test 2.7 so I do not know if the problem persists...

Ghostkeeper, 2017-09-11:
Just tried this on an UM3 (via wifi though, and on Ubuntu) and it just continued counting...

Perhaps the connection got lost somehow and Cura didn't notice, but just continued waiting for progress updates.

closed 2018-03-02 by ChrisTerBeke