Ultimaker/Cura #2336: CURA 2.6.2 Cost Calculation error

Geoloog, 2017-08-29:
There are cost calculation errors when calculation for different materials.

1.  If one has sliced a model with one type of material and then choose another type of material with a significant change in price the values stays the same.

2.  This calculation is true also in so far as that the cost is incorrectly calculated.  
eg.  I have a model that uses 4.73m of filament that cost R0.85/m.  The actual cost is R4.03 but Cura returns a value of R6.06

Regards,

Hendrik Bosman
Western Cape
South Africa

nallath, 2017-08-30:
I can't reproduce either of your issues on the master branch. If I switch to another material, the price changes accordingly. 
If you look at the Cost per Meter setting (which is automatically calculated) and multiply this by the length used, I get the correct value.

Geoloog, 2017-08-30:
Hi there Jaime,

Please check that attached photo.

Kindest regards

Hendrik Bosman

nallath, 2017-08-30:
Attachements need to be uploaded to github. If you attach them to the email, they get lost. 

Did you also check this with 2.7? This is the release that is closest to master.

Geoloog, 2017-08-30:
Hi Jaime,

Sorry did not know that.

Yes I did and it is the same problem.

Using length X cost as calculated where the material is configured differs
from the final amount provided by Cura.  Both 2.7 and 2.6.2

I also did a manual calculation by determining the g cost and multiplying
by g calculated by Cura the reply given by Cura is about 40% incorrect.

Kind regards,

Hendrik

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Hi   ,

*Kind regards,Hendrik*

*=======================================================NOTICE:  Be
Environmentally Responsible.Only print this email if absolutely unavoidable
!!!!!!!*

Geoloog, 2017-08-30:
[image]

I don`t know but Cura 2.7 & 2.6.2 use the cost values in "Materials and get R23.02 but manually calculated 18.38m = Cost R15.6230 and 54g = Cost R15.4165

Regards,

Hendrik
[image]

Geoloog, 2017-08-30:
Seem to have made another boo-boo :p 

Sorry

closed 2017-08-30 by Geoloog