Ultimaker/Cura #2338: UI Doesn't allow jerk values to be set less than 0.1

smartavionics, 2017-08-30:
I wanted to set travel jerk to 0 but the UI won't let me. All of the jerk settings have a min value of 0.1. Why can't the minimum values be 0?

nallath, 2017-08-30:
Because a jerk of 0 means that it's impossible to have an acceleration increase. Jerk is the acceleration of the acceleration. If you can't have an acceleration, you cant have a speed, as the starting speed is 0 (and with an acceleration of 0, it can't ever increase).

nallath, 2017-08-30:
Ugh. Turns out marlin does jerk differently, so in that case it does make sense to have jerk at 0 zero (even though physicly it still makes no sense to have a jerk of 0)

smartavionics, 2017-08-30:
For the printer I use, jerk is not not the derivative of acceleration. Instead it is taken to be the maximum instantaneous velocity change that the axis is capable of.

nallath, 2017-08-31:
Yeah, it's the same for Marlin. My confusion stemmed from the definition as used pretty much everywhere else.

Ghostkeeper, 2017-09-10:
What a jerk the guy that thought of that must've been :)

Ghostkeeper, 2017-09-10:
I just tried a jerk of 0mm/s on my Ultimaker 2 Extended+, which is based on Marlin. Works fine there. I'm going to reduce the minimum to 0 then.

smartavionics, 2017-09-10:
OK, thanks.

commit 32c5583a (2017-09-10): Reduce minimum jerk value to 0

A jerk of 0 is very possible, because Jerk in 3D printing doesn't mean the derivative of acceleration but rather the maximum instantaneous change in velocity on an axis.

Fixes #2338.
closed 2017-09-10 by Ghostkeeper