Ultimaker/Cura #2339: 2.6.2 Windows, printhead settings counted as model outline!

DIYglenn, 2017-08-30:
While my model fit in Cura on Mac, it doesn't on Windows. Same machine settings, model settings etc. I even copied the numbers to be absolutely sure. I get an error stating that my model is to large for the printbed, and it cannot Slice.

After messing around with scaling, I found the issue: When I scaled it down so much that the grey shadow indicating printhead movement was inside the printbed, it would slice. 

Seems like Cura Windows is doing some miscalculations there.

nallath, 2017-08-30:
Rounding issues can always occur. They can be slightly different depending on the architecture of your computer.

DIYglenn, 2017-08-30:
No, that's not the problem. The problem is that Cura adds the printhead min/max settings (usually for printing multiple objects one at a time) to the model dimensions. Meaning that I have to print at 1/3 scale or less to fit the model on the bed (because I have to fit the grey shadow indicating printhead movements inside the bed dimensions)

nallath, 2017-08-30:
I very much doubt that this is an issue that has anything to do with your OS. Are you very sure all settings are the same? It sounds like you made changes to the print sequence setting.

DIYglenn, 2017-08-30:
Yes, no differences at all. Most settings are default on both machine. Material profiles are imported, machine settings are exactly the same. 
I'll post a few screens when I'm at the office.

DIYglenn, 2017-08-31:
This is on Mac, notice the grey outline for the printhead movement:
[image]
Settings on the Mac:
[image]

DIYglenn, 2017-08-31:
Here's the same model on Windows, doesn't fit:
[image]
I have to scale it down until the grey outline is within the bed, THEN it will slice:
[image]
Settings on Windows:
[image]

nallath, 2017-08-31:
Did you also try it with a project file to transfer the settings? That way it's pretty much guaranteed that it's the same. 
Could you share that project file. Then we can also test & check what's going on.

DIYglenn, 2017-08-31:
Of course I can share, here:
[323_Test.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1266730/323_Test.curaproject.3mf.zip)

The project was however not displayed same on Windows and Mac (both 2.6.2). On the Windows computer it was further towards Y0, and I had to move it to center. Unfortunately not helping though, I still cannot get it to work on Windows. 

Would be interesting to see if you can replicate the issue. Let me know if you need more files/info.

nallath, 2017-09-01:
I've just tried it on linux, and I get the same results as you get on Mac.

LipuFei, 2017-09-01:
Hi @DIYglenn, were you already using Cura 2.5 on Windows when you have your custom printer?

LipuFei, 2017-09-01:
@DIYglenn Could also check if you are running Cura 2.5 on Windows. Because judging from your screenshot on Windows, there is no "number of extruder" setting in the machine configuration. This should only be the case in 2.5...

Ghostkeeper, 2017-09-10:
Slices fine on Cura 2.7 on Windows 8.1:

[image]

Ghostkeeper, 2017-09-10:
For some reason, in your Windows screenshot, it appears that something else is being printed through the other object? There is another shadow there at least.

DIYglenn, 2017-09-26:
You are right, I did not notice that before. But it is only one Object there. The Version was the exact same on Windows and Mac, and I installed them directly, no updates installed. 
I also reacted to the extruder setting missing, but I don't need that setting, so I didn't think about it. 

I haven't tried 2.7, yet, so that's promising! I finally just used my Mac for this print, but it's a model I'm still going to be working on, and our company is switching to PC.

Thanks :)

Fromen8923, 2017-11-15:
Have there been any fixes for this? I'm getting the same thing in Cura 3.0.4 on Windows 10. I only have this computer, so can't use the the above workaround.

I've done some experimenting, and the gray zones only appear when changing the print sequence option to "One at a time" (in case that wasn't clear in the above posts). I tried changing around the machines in Cura, and got some interesting results: the fake gray region is present with a Prusa i3 Mk1 (just called Prusa i3 in Cura's machine list), but goes away when using a Prusa i3 Mk2 and Ultimaker 2 (haven't tried other machines).

Prusa i3 Mk1 with fake region:
[image]

Prusa i3 Mk2 without fake region:
[image]

Another thing I've observed is when changing the printhead settings, the normal gray region around the object doesn't change size as expected. From what I can tell, it only extends out by the smaller of the values in a given dimension (Example: x-min:10 and x-max:50 would extend the gray region 10mm left and right of the object. Similar effect in y direction).

[image]

Surely this should not be symmetric, right? If anything, it should take the larger value. Another thing that happens is that the region sometimes refuses to display when the min and max values for an axis are the same (Example: x-min: 20, y-min:10, x-max:20, y-max:20). Very odd behavior.

[image]

I should also note that when using the Prusa i3 Mk1, only the normal gray region is affected by the printhead settings. The fake one remains the same size, and is offset a bit from the center of the model.

I've also been playing around with brims/skirts/rafts. Changing the distances these extend from the model affect both regions. Using brim as an example (and let's assume a print head of 20mm in each direction), setting the width to 0 (basically no brim) means that any object within 20mm of another object wont slice. Cura gives a message stating that none of the model fits in the build volume, which is to be expected because the print head would hit the first object as it prints the second object. So two objects 21mm apart would print just fine. If the brim width is increased to 5mm, the size of the gray region increases to 25mm, but both objects still slice even though the objects are within each other's gray regions. Moving them within 20mm causes them to not slice. So the graphic of the normal gray region is affected by the brim/skirt/raft, but not the actual boundaries.

[image]

It should also be noted that setting a brim to something large like 25mm causes extra lines to be printed on top of each other, which is clearly not desirable.

[image]

The fake region is also affected by brims/skirts/rafts, both functionally and graphically. Let's say it has a distance of 30mm from the object (don't actually know, and it's different in each direction) and 0 brim. If the object is 31mm from the edge, the Cura slices it. If the brim is increased to 5mm, then the fake region extends outside of the print bed, causing Cura to give a message about the model not fitting inside the build volume, so it doesn't slice.

That's all I've found so far. It would be great if someone could look into this more, as it's making my life difficult when printing multiple objects one at a time. Thanks!

nallath, 2017-11-15:
This behavior is intentional; When printing one at a time, there are suddenly 2 regions per object that cura needs to take into account; Areas where no model can ever be printed and areas where in some cases a model can not be printed (due to head size).
As most third party printers don't get a whole lot of time & effort put into their settings, they tend to have not quite correct head sizes (as those are only used for one at a time printing).

Ghostkeeper, 2017-11-24:
> Another thing I've observed is when changing the printhead settings, the normal gray region around the object doesn't change size as expected. From what I can tell, it only extends out by the [...]

You can always find an order of printing in which it fits just with the smaller collision offset. It's actually taking a mirror both horizontally and vertically (=180 degree rotation) and then taking the intersection of the original shape with that mirror. The proof for why this always works is kind of complex but it has been proven. It should be symmetric due to that intersection, but it's not always centred.

> I've also been playing around with brims/skirts/rafts. Changing the distances these extend from the model affect both regions.

There is a limitation in Cura here due to the way it's implemented. When you print one-at-a-time, Cura effectively does a complete slice for every model, including rafts, skirts and brims. As a result, it doesn't know the collision area that the platform adhesion of the previous parts incurred, can't merge the brims or skirts, the supports, and all that sort of thing. This is why we have to include the brim/skirt/raft size in the collision areas to prevent double extrusion in the intersection areas.

When I'm using the Prusa i3 (mk1), I'm only seeing one collision area:
[image]

ianpaschal, 2018-03-05:
Closing because this is no longer an issue in versions 3.x, and because this thread has received no activity for 3+ months.

closed 2018-03-05 by ianpaschal