Ultimaker/Cura #2344: Cura 2.7.0 Cancel print

Orlandoknox, 2017-08-30:
When cancelling a started print job the printer runs a z-calibration (on Prusa MK2), which causes a colition with the cancelled print (and potentially causes damage to the printhead).

nallath, 2017-08-31:
I'm not familliar with the Prusa machines, but this sounds like something that your printer does. Cura doesn't trigger any calibration as far as I'm aware.

Orlandoknox, 2017-08-31:
Just to be more precise: This happens when printing over USB and cancelling the print in the Cura software.  It is fine to cancel the print on the printer when started from SD card (the gcode is fine). So the problem could well be with the machine and I'll address it with Prusa.

Jhinta, 2017-09-01:
Duplicate of #2307

closed 2017-09-10 by Ghostkeeper