Ultimaker/Cura #2345: memory usage/leak in beta 2.7

MFHFozzy, 2017-08-30:
I have 32GB of RAM.
Windows 10 has complained and shut down Cura Beta 2.7 a few times now.
I *think* it happens when the camera stops updating, but i cant be sure, as Windows is killing the process before i can verify.

MFHFozzy, 2017-08-30:
Is there a way to stop/turn off  the camera?

MFHFozzy, 2017-08-31:
did another small 2h30min print. 
26min left, and Cura is at 17.4 GB of RAM, while using only ~2.5% of CPU....

fieldOfView, 2017-08-31:
Please try if the same thing happens if you switch back to the "prepare" screen

marcelk, 2017-09-01:
It seems that there is also a memory leak in the final version (2.7.0). I'm currently making my first print with this Cura version. About two hours after starting the print, my Mac started complaining about the memory consumption of Cura.

marcelk, 2017-09-01:
It's a few hours later now, and the memory consumption of Cura has increased to more than 6 GB. Switching back to the "prepare" screen doesn't free memory.

fieldOfView, 2017-09-01:
Note that you don't have to keep Cura open while printing with the UM3

marcelk, 2017-09-01:
That's true. But still, I don't like leaking programs, so I went back to the previous version.

fieldOfView, 2017-09-04:
In case it is the camera stream causing this:
https://github.com/Ultimaker/Cura/pull/2380

MFHFozzy, 2017-09-05:
Updated to 2.7 release.
Doing a small test print to test. (1.5hrs)
Leaving Cura on prepare screen, then minimizing program. 
Still ballooning up. @ 41min left, memory at 4GB. 
Restored Cura, switched to Print screen. Image stuck. Waited a few seconds, image becomes unstuck.
Memory size does not reduce.
Switch back to prepare screen, minimize. 
No change.

MFHFozzy, 2017-09-05:
More data:
Print finished. Memory was @ 17GB.

Still going up.
Currently @ 21.4GB.
Interestingly enough, the Print screen never updated past 84%, and still says "Printing".
The print is completely done, bed is cooled down, print removed.
Memory is now up to 22.7GB.

nallath, 2017-09-05:
I think I fixed this yesterday. As far as I could see, the issue only occured when you switched to the print tab, back to the buildplate and back to the print tab again.

MFHFozzy, 2017-09-06:
Let me know when there is a build to test.

MFHFozzy, 2017-11-04:
So far, this is fixed in 3.0. Closing.

closed 2017-11-04 by MFHFozzy