Ultimaker/Cura #2347: Can't Slice after changing printhead size settings

McAJBen, 2017-08-31:
When printing 2 models with "Print Sequence" set to "One at a Time" the slicer refuses to slice and hangs. The printhead outline is not shown and it seems like Cura can't load it.

version: Cura 2.7 (non-beta). I tried uninstalling and re-installing. Slicing multiple models worked before editing the printhead settings, and now does not work.

the following error is printed to the log

- ERROR - UM.Scene.SceneNode.callDecoration [239]: Exception calling decoration getConvexHullHead: could not convert string to float: '[[-30,60],[-30,-30],[60,60],[60,null]]'
- 2017-08-31 00:28:17,414 - ERROR - UM.Logger.logException [76]: Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x0000026599D32DA0> caused an exception
- 2017-08-31 00:28:17,417 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
- 2017-08-31 00:28:17,418 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 125, in run
- 2017-08-31 00:28:17,425 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.7\plugins\CuraEngineBackend\StartSliceJob.py", line 121, in run
- 2017-08-31 00:28:17,426 - ERROR - UM.Logger.logException [80]:     for node in OneAtATimeIterator(self._scene.getRoot()):
- 2017-08-31 00:28:17,428 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\OneAtATimeIterator.py", line 14, in __init__
- 2017-08-31 00:28:17,431 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Scene\Iterator\Iterator.py", line 11, in __init__
- 2017-08-31 00:28:17,432 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\OneAtATimeIterator.py", line 35, in _fillStack
- 2017-08-31 00:28:17,433 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\OneAtATimeIterator.py", line 35, in <listcomp>
- 2017-08-31 00:28:17,434 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\OneAtATimeIterator.py", line 35, in <listcomp>
- 2017-08-31 00:28:17,435 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\OneAtATimeIterator.py", line 95, in _checkHit
- 2017-08-31 00:28:17,436 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Math\Polygon.py", line 254, in intersectsPolygon
- 2017-08-31 00:28:17,438 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getPoints'

it seems getConvexHullHead is supposed to return 4 points for the printhead border, but it is returning "[[-30,60],[-30,-30],[60,60],[60,null]]" instead of "[[-30,60],[-30,-30],[60,60],[60,-30]]" causing it to crash.

[image]

McAJBen, 2017-08-31:
In <b>2.7/definition_changes/%PRINTER_NAME%_settings.inst.cfg</b> I changed

machine_head_with_fans_polygon = [[-30,60],[-30,-70],[60,60],[60,<b>null</b>]]

to

machine_head_with_fans_polygon = [[-30,60],[-30,-70],[60,60],[60,<b>-70</b>]]

its a temporary fix, but if I changed the printhead settings against it recreates a null in the array. Whatever code is creating machine_head_with_fans_polygon is creating a null.

nallath, 2017-08-31:
@fieldOfView Hmm. Seems that the machine actions is derping out somewhere. Is this something that we broke or has this been borked for a while?

fieldOfView, 2017-08-31:
I have not seen Machine Settings doing this. Is it reproducible?

McAJBen, 2017-08-31:
As far as I can tell, when you create a new printer it makes a _settings.inst.cfg file for it in definition_changes. Then when you ever edit one of the printhead settings it will append 

machine_head_with_fans_polygon = [[-X_min, Y_max],[-X_min, -Y_min],[X_max, Y_max],[X_max, <b>null</b>]]

This happens in version 2.7 but not 2.6.2

On this line here there is an accent over <b>mìn</b>, not sure why it would cause this problem now though.

https://github.com/Ultimaker/Cura/blob/2ac67a7d348edd0877875cca8fa5461bf471e33a/plugins/MachineSettingsAction/MachineSettingsAction.qml#L827

fieldOfView, 2017-08-31:
Good spot of debugging. Do you want to create a pull request or shall I?

McAJBen, 2017-08-31:
You can do it

commit 3c641759 (2017-09-01): Fix setting machine head polygon in Machine Settings

This typo was spotted by @McAJBen. Fixes #2347